raspberrypi/picamera2 #2: OpenCV 64-Bit install Script

meawoppl, 2022-02-17:
I saw your request in the readme, so I though I would share this recipe:
```bash
#!/bin/bash

# Requires the following apt packages to be installed:
# build-essential cmake git unzip pkg-config            \
# libjpeg-dev libtiff-dev libpng-dev libavcodec-dev libavformat-dev      \
# libswscale-dev libgtk2.0-dev libcanberra-gtk* libgtk-3-dev             \
# libgstreamer1.0-dev gstreamer1.0-gtk3 libgstreamer-plugins-base1.0-dev \
# gstreamer1.0-gl libxvidcore-dev libx264-dev python3-dev python3-numpy  \
# python3-pip libtbb2 libtbb-dev libdc1394-22-dev libv4l-dev v4l-utils   \
# libopenblas-dev libatlas-base-dev libblas-dev liblapack-dev gfortran   \
# libhdf5-dev libprotobuf-dev libgoogle-glog-dev libgflags-dev protobuf-compiler

set -e
echo "Installing OpenCV 4.5.5 on your Raspberry Pi 64-bit OS"
echo "This is going to take a while. Get a coffee/beer"
start_time=$(date +%s)

# download the latest version
cd ~
sudo rm -rf opencv*
wget -O opencv.zip https://github.com/opencv/opencv/archive/4.5.5.zip
wget -O opencv_contrib.zip https://github.com/opencv/opencv_contrib/archive/4.5.5.zip

# unpack
unzip opencv.zip
unzip opencv_contrib.zip

# some administration to make live easier later on
mv opencv-4.5.5 opencv
mv opencv_contrib-4.5.5 opencv_contrib

# clean up the zip files
rm opencv.zip
rm opencv_contrib.zip

# set install dir
mkdir ~/opencv/build
cd ~/opencv/build

# run cmake
cmake -D CMAKE_BUILD_TYPE=RELEASE \
-D CMAKE_INSTALL_PREFIX=/usr/local \
-D OPENCV_EXTRA_MODULES_PATH=~/opencv_contrib/modules \
-D ENABLE_NEON=ON \
-D WITH_OPENMP=ON \
-D WITH_OPENCL=OFF \
-D BUILD_TIFF=ON \
-D WITH_FFMPEG=ON \
-D WITH_TBB=ON \
-D BUILD_TBB=ON \
-D WITH_GSTREAMER=ON \
-D BUILD_TESTS=OFF \
-D WITH_EIGEN=OFF \
-D WITH_V4L=ON \
-D WITH_LIBV4L=ON \
-D WITH_VTK=OFF \
-D WITH_QT=OFF \
-D OPENCV_ENABLE_NONFREE=ON \
-D INSTALL_C_EXAMPLES=OFF \
-D INSTALL_PYTHON_EXAMPLES=OFF \
-D PYTHON3_PACKAGES_PATH=/usr/lib/python3/dist-packages \
-D OPENCV_GENERATE_PKGCONFIG=ON \
-D BUILD_EXAMPLES=OFF ..

# run make
make -j4
sudo make install
sudo ldconfig

# cleaning (frees 300 MB)
make clean

echo "Congratulations!"
echo "You've successfully installed OpenCV 4.5.5 on your Raspberry Pi 64-bit OS"
end_time=$(date +%s)
elapsed=$(( end_time - start_time ))

eval "echo Elapsed time: $(date -ud "@$elapsed" +'$((%s/3600/24)) days %H hr %M min %S sec')"
```

This takes quite a number of hours to run . . .

meawoppl, 2022-02-17:
How else can I help btw. I have a 3/4 finished Cython wrapping of libcamera that I would really maintain if I can get this working well. I am genuinely happy to plow about 60 hours of work into this.

aallan, 2022-02-17:
So @davidplowman is the person you need to talk to there!

davidplowman, 2022-02-17:
Thanks for that script. Perhaps we will make a folder for "useful bits and pieces" where we could keep such things, though that probably involves a commitment to maintain it. But I still wonder if there is a quick way to install OpenCV on a 64-bit image. For example, on a 32-bit image the version I used installs instantly, perhaps something like that will appear at some point?

As regards any kind of wrappers for libcamera, I would expect that we'll be sticking with the "official" bindings that will ship with libcamera (of which we are currently using an early version). If you've done something interesting is there anything you could propose to the libcamera mailing list?

kbingham, 2022-02-17:
Yes, if you have an implementation for binding libcamera to python, please share it on the libcamera mailing list.

meawoppl, 2022-02-18:
Here is the current progress:
```cython
import mmap

import tom.misc
from cython import NULL, size_t

from libcpp.vector cimport vector
from libcpp.string cimport string
from libcpp.memory cimport unique_ptr, shared_ptr
from libcpp cimport bool
from libc.stdint cimport uint32_t, uint64_t
from posix.unistd cimport close, read, off_t

cdef extern from "sys/types.h":
    ctypedef dev_t;

cdef extern from "libcamera/libcamera.h" namespace "libcamera":
    ctypedef enum ControlType:
        ControlTypeNone,
        ControlTypeBool,
        ControlTypeByte,
        ControlTypeInteger32,
        ControlTypeInteger64,
        ControlTypeFloat,
        ControlTypeString,
        ControlTypeRectangle,
        ControlTypeSize,

    ctypedef enum StreamRole:
        Raw,
        StillCapture,
        VideoRecording,
        Viewfinder

    ctypedef enum NoiseReductionModeEnum:
        NoiseReductionModeOff = 0,
        NoiseReductionModeFast = 1,
        NoiseReductionModeHighQuality = 2,
        NoiseReductionModeMinimal = 3,
        NoiseReductionModeZSL = 4,

    ctypedef enum CC_Status "libcamera::CameraConfiguration::Status":
        Valid "libcamera::CameraConfiguration::Valid"
        Adjusted "libcamera::CameraConfiguration::Adjusted"
        Invalid "libcamera::CameraConfiguration::Adjusted"

    cdef cppclass Private:
        pass

    cdef cppclass Size:
        unsigned int width;
        unsigned int height;

    cdef cppclass Stream:
        const StreamConfiguration &configuration();

    cdef cppclass ControlId:
        ControlId(unsigned int id, const string &name, ControlType type);
        unsigned int id();
        const string &name();

    cdef cppclass ControlValue:
        pass

    cdef cppclass ControlList:
        # void set(unsigned int id, const ControlValue &value);
        const ControlValue &get(unsigned int id) const;
        bool contains(unsigned int id) const;
        bool empty();
[... 217 lines cut ...]

        self.create_requests()

    def create_requests(self):
        self._dbg("Creating requests")
        self.request = self.camera.get().createRequest()
        assert self.request.get() is not NULL, "Failed to create request object?"
        self.request.get().addBuffer(self.stream_cfg.stream(), self.buffers.at(0))
        

    def _dbg(self, message):
        if self.debug:
            tom.misc.awesome(message)

    def __dealloc__(self):
        if self.allocator is not NULL:
            del self.allocator

        if self.camera.get():
            self.camera.get().release()
            self._dbg("Released Camera")

        self.cm.stop()  
        self._dbg("Stopped camera manager")
```

meawoppl, 2022-02-18:
I just signed up. Closing this issue, then picking up the conversation there.

davidplowman, 2022-02-19:
Hi again, I think perhaps @kbingham might be able to help you out with that.

pinchartl, 2022-02-20:
@meawoppl I've replied to your e-mail inquiry, could you check your spam folder ?

meawoppl, 2022-02-22:
It was indeed getting spam filtered. I am now subscribed and look forward to working with y'all

closed 2022-02-22 by meawoppl