raspberrypi/picamera2 #5: Compatibility with previous PiCamera python library

pageauc, 2022-02-18:
I have projects that will need to be migrated and was hoping the picamera2 python syntax once initiated will be compatible with the picamera syntax and recipes.  Can you provide some insight as to how much effort can be expected to migrate a project or make existing projects cross compatible. 
Thanks

davidplowman, 2022-02-19:
Hi, that's a good question but also a very broad one which makes it quite difficult to answer.

The first thing to be aware of is that Picamera2 is built on _completely_ different technologies to the original Picamera, namely standard Linux kernel drivers, a userspace framework running on the Arm cores with an API that is copied, pretty much directly, into Python. Picamera2's API is therefore slanted very much towards the features that libcamera offers, not so much to matching what Picamera used to do. But it also makes it possible for anyone to add or tweak features of Picamera2 where the underlying libcamera API allows it.

Just as an example, Picamera2 makes it easy to determine the exact camera settings of any particular image, rather than the settings that were in use "around the time of that image", so inevitably the data structures and APIs are going to reflect this. Picamera had some special means of creating overlays on camera images whereas Picamera2 will be using standard Linux frameworks (you can efficiently embed preview streams into PyQt applications, for example).

It's of course all going to vary on a case by case basis. Starting the camera and getting images should remain very straightforward, but applications that are tied into the existing Picamera API are going to have to change more. Part of the purpose of this release is to get feedback so that we discover such issues and seek to address them as best we can, so where people have specific use cases that appear to be problematic we'd like to learn more. Sorry that's such a vague answer!

davidplowman, 2022-05-19:
I think I'll probably close this one now, there hasn't been a lot of activity! There is always the Discussions page for more general discussions, and wherever there are specific issues (maybe "how do I do X in Picamera2" type stuff) those can certainly be raised here. Thanks!