raspberrypi/picamera2 #6: raw image functionality

hunkus, 2022-02-20:
1. I found "raw.py" example works just fine.

`picam2.capture_array("raw")`

but, the result of this function was **different** from the result of the **"picamera"**
```
stream = io.BytesIO()
camera.capture(stream, format='jpeg', bayer=True)
```
It also contains metadata of raw stream.
In this way, you can easily process the raw stream by passing them to a raw decoder,
for example,

`rawpy.imread(steram)`

2. Also, In the raw stream mode, the image **saving function is different** from the intended one.

```
config = picam2.preview_configuration(raw={"size": picam2.sensor_resolution})
preview = NullPreview(picam2)
picam2.configure(config)
picam2.start({"AwbMode": 6, "ExposureTime": 20000, "AnalogueGain": 1.0})
picam2.capture_file('test.jpg')
```

This does not save the raw stream, but the preview(640x480) image only.
I thought it would be saved the file like the following legacy method
(jpg+metadata+raw stream[after 'BCRM' string]).

`raspistill -r -o 'test.jpg'`

and the following command is **not implemented** and an error occurs.

`picam2.capture_file('test.dng')`

I thought it would be saved the file like the libcamera.

`libcamera-still -r -o test.dng`

davidplowman, 2022-02-20:
Hi, the behaviour of Picamera2 is different because the underlying camera stack is completely different.

You can capture raw image buffers and the metadata associated with them in various ways, but the following should work:
```
request = picam2.capture_request()
raw_image = request.make_array("raw")
metadata = request.get_metadata()
request.relase()
```
Saving a raw buffer as a JPEG makes no sense, indeed the legacy stack doesn't do it either, it merely appends the raw data in a non-standard way onto the end of a JPEG file. We haven't as yet added any specific functionality for saving DNG files, but when we do this it will be by using an existing Python module. I notice that the `rawpy` module doesn't support writing DNGs, but `PiDNG` (written by a Pi user, I believe) does (see [this example](https://github.com/schoolpost/PiDNG/blob/master/examples/raw2dng.py)). So this may be the kind of route we will follow.

hunkus, 2022-02-28:
Thank you very much for the detailed reply. Your answer has been very helpful.
I hope you will positively add saving function for DNGs you mentioned!

closed 2022-03-05 by hunkus