raspberrypi/picamera2 #7: Metering for White Balance

Wied58, 2022-02-20:
Could Picamera2 operate such that if white balance is supplied, the camera does not meter for it decreasing the over imaging run time? I volunteer at a Planetarium and we use the existing PiCamera to image light pollution. As we want consistency in the images, we set white balance. We have started to experiment with image durations of up to 30 seconds and those run times are quite lengthy.

davidplowman, 2022-02-20:
Yes, when you do `picam2.start()` you can instead do this `picam2.start({'ColourGains': (1.5, 2.5)})`. Clearly you can use whatever fixed values you like in place of 1.5 and 2.5. If the camera is running and you want to query the current colour gains, you can do `picam2.capture_metadata()['ColourGains']`. You can also fix the exposure time (`'ExposureTime'`) and gain (`'AnalogueGain'`) if you want.

Wied58, 2022-02-21:
Thank you,

Will manually setting the gains, stop PiCamera2 from taking the time to determine them?

davidplowman, 2022-02-21:
> Will manually setting the gains, stop PiCamera2 from taking the time to determine them?

Yes, that's correct.

davidplowman, 2022-03-02:
Please feel free to open another issue if you have more questions. Thanks!

closed 2022-03-02 by davidplowman