raspberrypi/picamera2 #11: v4l2 module missing in installation instructions

tvoverbeek, 2022-02-23:
With the addition of the video files, picamera2.py now imports encoder.py which in turn imports v4l2.
However python module v4l2 is not available in apt (at least I could not find it).
There is a v4l2 python module on Pypi.
Is that the one we need?

meawoppl, 2022-02-23:
There is some language in the readme about it, do you have a suggestion for how this section can be a bit more clear/easy to find?
https://github.com/raspberrypi/picamera2/blob/main/README.md?plain=1#L37

tvoverbeek, 2022-02-23:
I missed todays README update.

v4l2 is not only needed for video encoding, but also for anything which imports picamera2.
Just running the preview.py example aborts when the v4l2 module is not found.
picamera2.py imports encoder.py which istarts with `from v4l2 import *`
Suggested text:
```
We also need the python-v4l2 module. The official version is hopelessly out of date and appears to be
unmaintained for many years, so please install the fork below:
```
Note: Todays Tensorflow blog post is also out of date.

davidplowman, 2022-02-23:
Thanks for pointing that out. Unfortunately this whole thing is a bit of a moving target at the moment so it's going to be a bit tricky from time to time. I'll see about trying to get some of those things updated, and eventually there'll be "proper" documentation rather than a somewhat unofficial readme file. Hopefully things will stop changing quite so much soon and the situation will improve.

closed 2022-02-24 by tvoverbeek