raspberrypi/picamera2 #13: Personal Hitlist

meawoppl, 2022-02-23:
I am excited to contribute a fair bit of effort to getting this repository into more usable state. I currently rely on it for my day-job, and I am eager to make it better.

Here is the list of things, in rough order of my perceived importance, that I was thinking to work on:
* Turning this repository into a more normally structured python package
* Document what is needed for the various preview interfaces (I am mainly interested in headless operations, so I will need help here)
* Make some dependency installer scripts/support
* Get the package up on pypi.
* Start working on a backward compatibility wrapper - Filling the feature gaps between this and the original `picamera` module.

meawoppl, 2022-02-23:
Any feedback welcome!

davidplowman, 2022-02-23:
I think most of what you want - proper installation etc. - goes without saying. This is an early preview release just to give folks a chance to try stuff, it's obviously not going to stay like that. As regards a "backwards compatibility wrapper" - there are things that worked in the old world that basically won't work, or at least can't work in the same way in the new world, and vice versa. If more specific examples could be given, that would clearly help us to know what things to look at.

meawoppl, 2022-02-23:
Totally agree. There is a lot of stuff in `picamera` that just does not map well onto the `libcamera` stack. That said, there are some very basic bits that do, that i suspect would cover some use cases.

If I tossed together a quick packaging PR would you consider it?

IanTBlack, 2022-02-23:
I just started messing around with Picamera2 and have been periodically looking at the PRs and issues. Would it be possible to enable the Discussions page so it might be easier to track issues against feature requests?

davidplowman, 2022-02-24:
@meawoppl I'm not really a Python programmer so yes, it might be helpful to have a packaging PR. As I keep reminding everyone, it's very early days, but reducing the installation hurdle will help more people to give it a try.

@IanTBlack Thanks for the suggestion, I've enabled that. I've not used GitHub Discussions before, but anything that helps break rather general discussions into more specific topics could be helpful, I think.

meawoppl, 2022-02-24:
Totally understood wrt. to the early days comment. But also willing to pitch in and shovel making this a good thing :)

meawoppl, 2022-02-24:
I added a bullet to the above list wrt to resource cleanup.

davidplowman, 2022-05-19:
Hi, I think I'm going to close this one. There's not been too much going on here and issues which are actually lists of stuff are quite hard to deal with. Actually quite a few of the things here have improved in any case (with your help in some cases!!), and we can continue to file single-item-issues here or PRs as we go, as we have in fact been doing in practice. Thanks!