raspberrypi/picamera2 #18: opencv_mertens_merge example hangs when using v1 camera

tvoverbeek, 2022-02-24:
For details see the (closed) PR #15

tvoverbeek, 2022-03-02:
@davidplowman  I ran the opencv_mertens_merge example with LIBCAMERA_LOG_LEVEL defines as *:DEBUG.
Produced tons of output. One thing I noticed at the point where it goes off the rails with the V1 camera is the 4 buffer ids got completely bogus (something in the 18000 range).
Does that help or give you a clue?
If you want I can run it again and post the debug output. Maybe you can suggest a more reasonable set of debug options instead of brute force everything..

davidplowman, 2022-03-07:
I think we've fixed this one, though the problems were rather deeper and more far-reaching than we had expected. I've pushed some updates to the picamera2 branch in our raspberrypi/libcamera repository so if you pull those and rebuild then things should work better.

tvoverbeek, 2022-03-07:
@davidplowman I rebuilt libcamera to include your fixes. I noticed the need for a kernel fix just now.
Most examples work, but the opencv-mertens-merge still fails with an error:
[1:08:14.234956117] [11296] ERROR V4L2 v4l2_videodevice.cpp:998 /dev/video14[14:cap]: Unable to set format: Invalid argument
After this all examples fail when trying to configure for preview with the same v4l2 error.
No way to recover except reboot.
So I am waiting for rpi-update (both 32-bit and 64-bit?) before trying again.

tvoverbeek, 2022-03-08:
@davidplowman I assume the needed kernel fix is the one in https://github.com/raspberrypi/linux/pull/4921.
This is now in the 5.15.26 kernel commit https://github.com/raspberrypi/rpi-firmware/commit/25147f6e86cad2e496e4e9dca4e7dfc87b261cab.
So I took a chance and ran rpi-update.
Kernel is now `Linux picam2-32 5.15.26-v7l+ #1528 SMP Fri Mar 4 15:42:30 GMT 2022 armv7l`
Result: opencv_mertens_merge runs without probems with the V1 camera.
Thanks for all the bug hunting and fix.
Closing the issue.

closed 2022-03-08 by tvoverbeek

tvoverbeek, 2022-03-08:
Additional info:
Previous was on Bullseye 32-bit.Also tried on 64-bit.
This time did the rpi-update to 5.15.26 first.
The mertens example still showed the same hang (as expected).
After rebuilding libcamera the mertens example worked fine.
So both 32-bit and 64-bit work now.