raspberrypi/picamera2 #22: No module named 'libcamera'

rogergithubbing, 2022-03-06:
Hello,

Thanks for your help; I'm sure that my doubt is because I'm a noob in raspberry, but I cannot find the answer googling it:

I'm using the latest version of raspbian 64 bits (Bullseye), installed two days ago...

I have followed all the installation steps for picamera2, and as far as I know, I had no problems.

I'm trying to test it with a simple .py , that only has a single line "import picamera2", but it complains that it cannot find module libcamera:

a2test1.py

Traceback (most recent call last):
  File "/home/pi/PycharmProjects/FotoMoniTests/Picamera2 tests/picamera2test1.py", line 3, in <module>
    import picamera2
  File "/home/pi/picamera2/picamera2.py", line 4, in <module>
    import libcamera
ModuleNotFoundError: No module named 'libcamera'

I don't know where libcamera is located, but I think it is on the system, because its tools are working correctly (for example, libcamera-hello shows the preview).

Thanks in advance,

Roger

rogergithubbing, 2022-03-06:
I'm sorry... I think I missed part of the initial instructions: I'll install the missing parts asap... thanks

rogergithubbing, 2022-03-06:
Hello again,

I've tested the library after installing and building libcamera applying the two differences you are mentioning in the instructions, and now I've got this error (in **bold**):

pi@raspberrypi:~ $ pwd
/home/pi
pi@raspberrypi:~ $ python3
Python 3.9.2 (default, Feb 28 2021, 17:03:44) 
[GCC 10.2.1 20210110] on linux
Type "help", "copyright", "credits" or "license" for more information.
import sys
sys.path
['', '/home/pi/picamera2', '/home/pi/libcamera/build/src/py', '/home/pi/kmsxx/build/py', '/home/pi/python-v4l2', '/usr/lib/python39.zip', '/usr/lib/python3.9', '/usr/lib/python3.9/lib-dynload', '/home/pi/.local/lib/python3.9/site-packages', '/usr/local/lib/python3.9/dist-packages', '/usr/lib/python3/dist-packages', '/usr/lib/python3.9/dist-packages']
exit()
pi@raspberrypi:~ $ python3 mytestpicamera2.py

**Traceback (most recent call last):
  File "/home/pi/mytestpicamera2.py", line 1, in <module>
    from picamera2 import *
  File "/home/pi/picamera2/picamera2.py", line 4, in <module>
    import libcamera
  File "/home/pi/libcamera/build/src/py/libcamera/__init__.py", line 4, in <module>
    from ._libcamera import *
ModuleNotFoundError: No module named 'libcamera._libcamera'**

davidplowman, 2022-03-07:
Hi, I'm not really sure what to suggest - perhaps the process of going back and re-doing bits of the build has left something in a bad state? I think I would be inclined to delete your libcamera folder, then check it out again from the raspberrypi/libcamera repo, and rebuild it from scratch. After the `meson` command, check the output to be sure that it reports "YES" for the "Python bindings".

apage224, 2022-03-11:
I ran into a similar error ```ModuleNotFoundError: No module named 'libcamera._libcamera'``` For me, the problem was that ```ninja -C build install``` failed for libcamera- you might need to limit parallelization with ```ninja -C build -j2```. I'd double check that libcamera fully built and installed. Contents of ```libcamera/build/src/py``` should look like following:

```

└── libcamera
    ├── __init__.py -> ../../../../src/py/libcamera/__init__.py
    ├── __pycache__
    │   └── __init__.cpython-39.pyc
    ├── _libcamera.so
    └── _libcamera.so.p
        └── pymain.cpp.o

3 directories, 4 files
```

rogergithubbing, 2022-03-12:
Thanks to everyone ! I have tried (without deleting the folder libcamera first) to rebuild libcamera with "ninja -C build -j2"; Now, the import succeeds, but when I try this script, it fails:

```
from picamera2 import *
picam2 = Picamera2()
```
I've noticed that some things are still missing from the tree structure @apage224 mentioned, even after issuing the new ninja command: I only have the subfolder "_libcamera.so.p" and the file "__init__.py".

This is the new error I'm getting:

> 
> [0:39:07.208709779] [4557]  INFO IPAManager ipa_manager.cpp:138 libcamera is not installed. Adding '/home/pi/libcamera/build/src/ipa' to the IPA search path
> [0:39:07.209461942] [4557]  INFO Camera camera_manager.cpp:293 libcamera v0.0.0+3421-16d99c30
> [0:39:07.218061298] [4564]  INFO IPAProxy ipa_proxy.cpp:130 libcamera is not installed. Loading IPA configuration from '/home/pi/libcamera/src/ipa/raspberrypi/data'
> [0:39:07.228360737] [4564]  WARN RPI raspberrypi.cpp:1216 Mismatch between Unicam and CamHelper for embedded data usage!
> [0:39:07.229523174] [4564]  INFO RPI raspberrypi.cpp:1331 Registered camera /base/soc/i2c0mux/i2c@1/imx219@10 to Unicam device /dev/media1 and ISP device /dev/media2
> Camera manager: <libcamera._libcamera.CameraManager object at 0x7fabca99f0>
> Made <picamera2.Picamera2 object at 0x7fabca5e80>
> Opened camera: <libcamera.Camera '/base/soc/i2c0mux/i2c@1/imx219@10'>
>

davidplowman, 2022-03-12:
To be honest, libcamera is a bit noisy about warnings when it starts, so none of this is actually an error. Are things working otherwise for you?

rogergithubbing, 2022-03-16:
> To be honest, libcamera is a bit noisy about warnings when it starts, so none of this is actually an error. Are things working otherwise for you?

Well,... Finally, I have deleted the libcamera folder and rebuild it again... Now what happens, is that the following code fails:

```
from picamera2.picamera2 import *
import time

picam2 = Picamera2()
picam2.start_preview(Preview.QTGL)

preview_config = picam2.preview_configuration()
picam2.configure(preview_config)

picam2.start()
time.sleep(5)
```

> picam2.start_preview(Preview.QTGL)
AttributeError: 'Picamera2' object has no attribute 'start_preview'

Thanks in advance !

rogergithubbing, 2022-03-16:
> There has been a big repository update just recently, so it seems like something has been left behind. Can you have a look in the picamera2/picamera2/picamera2.py file and see if there's a start_preview method in there? It should be [here](https://github.com/raspberrypi/picamera2/blob/main/picamera2/picamera2.py#L126).

That was it ! it was not present; I re-issued "git clone https://github.com/raspberrypi/picamera2.git" and now it works...

THANKS A LOT !

By the way... if I try to run this program connected from my Windows computer (with xrdp), it fails... I am supposing that the multimedia capabilities are not available for the camera connected to rdp, and for that, there is no workaround...:

> libEGL warning: DRI2: failed to authenticate
> EGL Mesa Project 1.5
> Exception in thread Thread-1:
> Traceback (most recent call last):
>   File "/usr/lib/python3.9/threading.py", line 954, in _bootstrap_inner
>     self.run()
>   File "/usr/lib/python3.9/threading.py", line 892, in run
>     self._target(*self._args, **self._kwargs)
>   File "/home/pi/picamera2/picamera2/previews/qt_gl_preview.py", line 13, in thread_func
>     self.qpicamera2 = QGlPicamera2(picam2, width=width, height=height)
>   File "/home/pi/picamera2/picamera2/previews/q_gl_picamera2.py", line 91, in __init__
>     self.egl = EglState()
>   File "/home/pi/picamera2/picamera2/previews/q_gl_picamera2.py", line 32, in __init__
>     self.choose_config()
>   File "/home/pi/picamera2/picamera2/previews/q_gl_picamera2.py", line 49, in choose_config
>     check_egl_extensions(self.display, ["EGL_EXT_image_dma_buf_import"])
>   File "/home/pi/picamera2/picamera2/previews/gl_helpers.py", line 70, in check_egl_extensions
>     raise Exception(ext + " missing")
> Exception: EGL_EXT_image_dma_buf_import missing
>

rogergithubbing, 2022-03-16:
> Try the preview_x_forwarding.py example. If your Windows PC speaks regular X Windows then it should work (the OpenGL accelerated preview implementation doesn't work with X forwarding). It uses regular software rendering, so will eat your CPU (relatively speaking).

That worked like a charm ! Thanks for your excellent advices.

Roger

closed 2022-03-16 by rogergithubbing