raspberrypi/picamera2 #38: No "sport" exposure mode

felixhesshaw, 2022-03-16:
Hello, 

thanks for your great work! I want to use the pi cam hq with the shortest shutter time possible.
Is there any way to activate the sports exposure mode with picamera2? 
This is from the libcamera -help output:
'--exposure arg (=normal)              Set the exposure mode (normal, sport)'

Couldn't find this option in picamera2 but maybe I missed something.

Thanks in advance
Felix

davidplowman, 2022-03-17:
I've pushed an update to the picamera2 branch of the raspberrypi/libcamera repo. If you rebuild this you should be able to use (for example) `libcamera.AeExposureMode.Short`.

davidplowman, 2022-03-18:
The fix was confirmed in another similar report so I'll close this one now. Please feel free to file a new report if you're still having problems. Thanks!

closed 2022-03-18 by davidplowman