raspberrypi/picamera2 #44: flash_mode

rogergithubbing, 2022-03-16:
Hello,

With the previous version I used "flash_mode" on with a LED to use it as a flash; How can I do it now ?

Thanks !

rogergithubbing, 2022-03-16:
And by the way, I used also overlays... is there a way to do it now ? thanks again

davidplowman, 2022-03-17:
The flash LED only works on the v1 camera, I believe. It's connected to one of the GPIO pins, though only for some Pis. The pin number may vary too. Maybe something to ask on the [Camera forum](https://forums.raspberrypi.com/viewforum.php?f=43&sid=247c8574ee295cf507f7c29e29e20a3f).

As regards overlays, that's something on the list of things to look at. The answer will be "whatever it is that you usually do for the GUI framework in question", but I can't really be more specific at the moment. For the software (non-OpenGL) Qt preview window you would probably find information out on the net, for the others it might be more difficult. It would be nice to add some kind of common API that works across the different types of preview window, but that's very much TBD.

closed 2022-05-19 by davidplowman