raspberrypi/picamera2 #48: Full resolution encoding doesn't work for camera v2

sdegrace, 2022-03-21:
```
#!/usr/bin/python3

from picamera2.encoders.h264_encoder import *
from picamera2.picamera2 import *
import time
import os

picam2 = Picamera2()
video_config = picam2.video_configuration({'size':picam2.sensor_resolution})
picam2.configure(video_config)

picam2.start_preview()
encoder = H264Encoder(10000000)

picam2.start_recording(encoder, 'test.h264')
time.sleep(10)
picam2.stop_recording()
```

gives the output 

```[0:01:59.761323091] [1703]  INFO IPAManager ipa_manager.cpp:138 libcamera is not installed. Adding '/home/pi/libcamera/build/src/ipa' to the IPA search path
[0:01:59.761992574] [1703]  INFO Camera camera_manager.cpp:293 libcamera v0.0.0+3447-c6bb050a
[0:01:59.770978897] [1704]  INFO IPAProxy ipa_proxy.cpp:130 libcamera is not installed. Loading IPA configuration from '/home/pi/libcamera/src/ipa/raspberrypi/data'
[0:01:59.785928705] [1704]  WARN RPI raspberrypi.cpp:1222 Mismatch between Unicam and CamHelper for embedded data usage!
[0:01:59.786501034] [1704]  INFO RPI raspberrypi.cpp:1337 Registered camera /base/soc/i2c0mux/i2c@1/imx219@10 to Unicam device /dev/media3 and ISP device /dev/media1
2022-03-21T15:51:31.784Z | INFO     | Initialization successful.
2022-03-21T15:51:31.786Z | INFO     | Camera now open.
[0:01:59.790679252] [1703]  INFO Camera camera.cpp:1028 configuring streams: (0) 3280x2464-XBGR8888
[0:01:59.791085052] [1704]  INFO RPI raspberrypi.cpp:757 Sensor: /base/soc/i2c0mux/i2c@1/imx219@10 - Selected sensor format: 3280x2464-SBGGR10_1X10 - Selected unicam format: 3280x2464-pBAA
2022-03-21T15:51:31.794Z | INFO     | Configuration successful!
Traceback (most recent call last):
  File "/home/pi/projects/Telescope/scratch.py", line 15, in <module>
    picam2.start_recording(encoder, 'test.h264')
  File "/home/pi/picamera2/picamera2/picamera2.py", line 856, in start_recording
    self.start_encoder()
  File "/home/pi/picamera2/picamera2/picamera2.py", line 834, in start_encoder
    self.encoder._start()
  File "/home/pi/picamera2/picamera2/encoders/h264_encoder.py", line 97, in _start
    fcntl.ioctl(self.vd, VIDIOC_STREAMON, typev)
ProcessLookupError: [Errno 3] No such process
```

H.264 obviously has some issues with 4k encoding, but my understanding is that it should be possible.

davidplowman, 2022-03-21:
Hi, yes, on the Raspberry Pi H.264 encoding is limited to 1080p. Actually you can go very slightly larger, I don't have the exact numbers in my head, but it's not much. We could probably improve the error message, though. Sorry about that.

sdegrace, 2022-03-21:
Yeah that's the impression I've been getting. Are there or will there be any options for higher resolutions? The highest resolution possible is critical for my use case.

davidplowman, 2022-03-22:
The approx-1080p limit for the h.264 encoder is down to the hardware and is, unfortunately, very fundamental. There are line buffers that are 2048 pixels wide so anything wider than that (in fact, exactly that size too) will fail. There is no prospect of any fixes or workarounds.

You could go larger if you encode using MJPEG, on a Pi 4 I'd expect you to achieve "several frames per second", though note that nothing will ever go faster than 10fps in full resolution because that's the limit at which we can read from the sensor.

Can you say what resolution you require? For example you could encode 1080p (or slightly larger) which is downscaled from the full resolution. That would still be limited to 10fps (because of the sensor readout), but you'd get better per-pixel detail than the default setup, though the latter would give you better framerates. So there are swings and roundabouts here...

sdegrace, 2022-03-22:
My application is astrophotography without tracking, so the answer is always "bigger". The picam2 is a remarkably good choice because its sensor is so small, which is a very good thing for astrophotography. I've just stumbled on the "digital zoom" example though, which may be a good fit, though it's definitely not going to be easy to set it up to track the chosen object. On the other hand, I'll be able to dump more images much more quickly than I'd ever be able to with a full resolution image, so it may be worth the struggle.

BTW I'm using picamera2 for its ability to control analogue gain, which is a HUGE selling point. Thanks so much for helping to put all this together!

davidplowman, 2022-03-30:
I think we've explained the original issue so I'll close this now. But if you have any other questions please feel free to open a new issue here instead! Thanks.

closed 2022-03-30 by davidplowman