raspberrypi/picamera2 #51: Get/Set Brightness etc.

barthm1, 2022-03-27:
How can I get/set brightness etc.?

davidplowman, 2022-03-28:
Hi, could you explain more precisely what you mean by "brightness"? _libcamera_ does offer a "brightness" control which specifies a fixed offset (positive or negative) that is added to all the output pixels. It's scaled so that 1.0 makes (just about) everything bright white, and -1.0 makes (nearly) everything black, with 0.0  being the default setting. You can set it like this:
```
picam2.set_controls({"brightness": 0.1})  # make everything brighter
picam2.set_controls({"brightness": -0.1})  # make everything darker
picam2.set_controls({"brightness": 0})  # default setting
```
We don't currently report the brightness value back because it only changes when the user requests it to change, though we are wondering a little bit whether this is the best behaviour or not.

There is also a "contrast" control (range of 0.0 to 32.0, where 0.0 makes everything mid-grey and 1.0 is "normal" contrast", larger values increase the contrast making pixels go increasingly towards black or white).

Finally there is an "ExposureValue" control which actually changes the exposure target value and will cause the AGC/AEC to converge to different shutter speeds and gains (the brightness and contrast controls have no effect on the AGC/AEC). This is in units of "stops", and the default value is 0.

barthm1, 2022-03-28:
Yes - this is what I mean brightness/contrast and saturation. At the moment I try to port this project
[https://github.com/Billwilliams1952/PiCameraApp](https://github.com/Billwilliams1952/PiCameraApp)
to picamera2

closed 2022-03-28 by barthm1

davidplowman, 2022-03-28:
Glad to help. Perhaps I should add there's also a "Saturation" control where 1.0 = normal colour saturation (the default), 0.0 makes it greyscale and > 1.0 increases the colour saturation. Similarly there's also "Sharpness" (0.0 = no sharpening, 1.0 = default, > 1.0 increases the sharpening).