raspberrypi/picamera2 #56: Manually Set Framerate

SamM-Springbok, 2022-04-01:
Libcamera allows a camera's framerate to be manually set/requested if compatible. I can't find anything in the current picamera2 documentation about how to set the framerate, and it seems to be automatically set based on the camera and resolution.

Folling https://github.com/raspberrypi/picamera2/blob/main/examples/controls.py and trying to set `FrameDuration` results in: `RuntimeError: Control FrameDuration not found`

davidplowman, 2022-04-03:
You should be able to use
```
picam2.start({"FrameDurationLimits": (33333, 33333)})
```
(or you can set it after starting the camera with `set_controls`). The two numbers are the minimum and maximum allowed frame durations in microseconds. Or you can make them the same to fix the framerate completely, as per the example (to 30fps).

davidplowman, 2022-04-06:
I think this issue should be resolved, so I'll go ahead and close it. However, please post back if there are any further problems. Thanks!

closed 2022-04-06 by davidplowman