raspberrypi/picamera2 #66: Camera cannot be used from a threaded context

sdegrace, 2022-04-13:
```
from picamera2.picamera2 import *

import threading

picam2 = Picamera2()
preview_config = picam2.preview_configuration()
picam2.configure(preview_config)

def worker():
	print('capturing')
	picam2.capture_file("test.jpg")
	print('captured and saved')

picam2.start()

thread = threading.Thread(target=worker)
thread.start()
print('thread started')
thread.join()
print('thread_completed')
```

Above is a minimal example of the problem. The problem remains if the camera is passed in as an argument and when it is started from inside the threaded context.

davidplowman, 2022-04-13:
Hi, thanks for reporting this. As far as I can see you may not be starting a "preview", which is really an event loop that drives the camera (see [here](https://github.com/raspberrypi/picamera2#driving-the-camera)). So maybe put `picam2.start_preview()` just after the `picam2.start()`?

sdegrace, 2022-04-13:
Yep. I promise I've read through that whole page like 10 times now, and still forgot about that.

I do have to say that for contexts where having a 'preview window' is entirely unnecessary, it's a surprise that you need to have one, even if it is the "null" preview. Even the concept of a "null preview" is pretty confusing. Personally I think that's a bit of a design failure. Perhaps there's a way to create a null preview by default when the camera is started?

davidplowman, 2022-04-14:
As I've said previously, things are very much still under review at this point. The biggest question in my mind relates to how we use Qt. Qt always runs its own event loop (and is annoyingly picky about how and where it runs), so the question is whether to piggy-back on that, or run another which would then have to poke the Qt one.

Currently we're doing the former, which means that if you don't start a Qt window then someone else must run an event loop. But we could run our own and work out how to deal with Qt, or we could mandate that you start your preview window _before_ `picam2.start()` so that we'd know to start our own, or we could make it always start both (are there any possible use cases that would be made more difficult by that?). Many questions, which is why I've opened up this "preview version" to get some experience and feedback.

As for this particular issue, I think it's been answered so I'll go ahead and close it.

closed 2022-04-14 by davidplowman

sdegrace, 2022-04-14:
Fully understood about this being a preview release, and as I've said before - thanks for your effort. I only meant it as a suggestion for how I think the final product should look, not how I expect it to be right now.

That said, if my opinion matters at all, I think that useful code should fail fast and loud. The easiest solution in my mind is just to throw an informative error or at least a warning when you try to capture a request, but there's no preview window started. I can easily throw together a PR for that.

I think the best solution is probably the one you mention - you must start your preview window before calling `start`. If it's a requirement for using the camera, why would it not be required before you start using it?

Just my two cents. I'd be happy to put something together for either.

davidplowman, 2022-04-15:
Hi, thanks for the suggestion. In the first instance I'm still quite keen to see if the whole thing can be run in a thread separate from the Qt one as that could have some other advantages. But obviously that's all for after the Easter break now.