raspberrypi/picamera2 #67: Qt widget based camera crashes after taking some stills

JamesRayn, 2022-04-13:
Hi,

I expirianced some crashes with Qt widget based camera preview (q_gl_picamera2) taking stills with switch_mode_and_capture_file.
Long terms short, I'm able to replicate the chrashes after taking some stills in the example "app_capture.py" too. Simply take some stills repeatedly and it will crash at around ten pics (at me the first try 8 times, the second try 13 times).
The error displayed will be: Camera did not start properly.

davidplowman, 2022-04-13:
Hi, thanks for reporting this, I'll look at it once I'm back in the office. Could you maybe clarify one or two further details?

- Let's check your operating system, what do `uname -a` and `vcgencmd version` report?
- What kind of device are we talking about. A Pi Zero, a Pi 4? The output of `cat /proc/cpuinfo` would be helpful.
- The output of `cat /proc/meminfo` is also useful.
- Posting the console output including the last successful capture up to and including the error would be helpful.
- Is there any particular pattern to the way you capture the shots. Are you trying to do them quickly, or does it not matter?

Thanks!

JamesRayn, 2022-04-16:
Hi, in my further experiments I discovered, it seems to occur in context of switching camera modes. The timing between the shots or the mode switching seems to be irrelevant, it occurs after some switches, at the app_capture.py example it occurs after around 10 shots. In my script I spitted the functions of mode changing, capturing image and saving the image to make it more responsive and for better timing of the different steps (by the way an example of a broke down switch_mode_and_capture_file could be nice) but in this configuration the error occurs often after only two mode changes. I'm using a Pi 4B 4GB with a PiCam HQ, I seted the VRAM to 256MB (because I tested around with the legacy picamera-lib before).

uname -a
Linux fotobox 5.15.32-v7l+ #1538 SMP Thu Mar 31 19:39:41 BST 2022 armv7l GNU/Linux

__________________________________________________________________________________________________________________

vcgencmd version
Mar 24 2022 13:19:26
Copyright (c) 2012 Broadcom
version e5a963efa66a1974127860b42e913d2374139ff5 (clean) (release) (start)

__________________________________________________________________________________________________________________

cat /proc/cpuinfo
processor       : 0
model name      : ARMv7 Processor rev 3 (v7l)
BogoMIPS        : 108.00
Features        : half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xd08
CPU revision    : 3

processor       : 1
model name      : ARMv7 Processor rev 3 (v7l)
BogoMIPS        : 108.00
Features        : half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xd08
CPU revision    : 3

processor       : 2
model name      : ARMv7 Processor rev 3 (v7l)
BogoMIPS        : 108.00
Features        : half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xd08
CPU revision    : 3

processor       : 3
model name      : ARMv7 Processor rev 3 (v7l)
BogoMIPS        : 108.00
Features        : half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xd08
CPU revision    : 3

Hardware        : BCM2711
Revision        : c03114
Serial          : 1000000003b38041
Model           : Raspberry Pi 4 Model B Rev 1.4

__________________________________________________________________________________________________________________

cat /proc/meminfo (with loaded app_cpture example)
MemTotal:        3748348 kB
MemFree:         3212620 kB
MemAvailable:    3418752 kB
Buffers:           24940 kB
Cached:           316440 kB
SwapCached:            0 kB
Active:           176588 kB
[... 56 lines cut ...]
[0:54:43.072383421] [1738]  INFO Camera camera.cpp:1028 configuring streams: (0) 800x600-XBGR8888
[0:54:43.093702329] [1739]  INFO RPI raspberrypi.cpp:757 Sensor: /base/soc/i2c0mux/i2c@1/imx477@1a - Selected sensor format: 2028x1520-SBGGR12_1X12 - Selected unicam format: 2028x1520-pBCC
2022-04-16T17:50:51.088Z | INFO     | Configuration successful!
[0:54:43.129335584] [1793]  INFO IPARPI raspberrypi.cpp:631 Request ctrl: NoiseReductionMode = 3
2022-04-16T17:50:51.631Z | INFO     | Camera started
2022-04-16T17:50:52.489Z | INFO     | Camera stopped
[0:54:44.499548916] [1738]  INFO Camera camera.cpp:1028 configuring streams: (0) 4048x3040-XBGR8888
[0:54:44.504859076] [1739]  INFO RPI raspberrypi.cpp:757 Sensor: /base/soc/i2c0mux/i2c@1/imx477@1a - Selected sensor format: 4056x3040-SBGGR12_1X12 - Selected unicam format: 4056x3040-pBCC
2022-04-16T17:50:52.500Z | INFO     | Configuration successful!
[0:54:44.671685742] [1739] ERROR V4L2 v4l2_videodevice.cpp:1192 /dev/video21[15:cap]: Unable to request 1 buffers: Cannot allocate memory
[0:54:44.671801766] [1739] ERROR RPI raspberrypi.cpp:1001 Failed to allocate buffers
2022-04-16T17:50:52.686Z | ERROR    | Camera did not start properly.
Traceback (most recent call last):
  File "/home/pi/picamera2/picamera2/previews/q_gl_picamera2.py", line 316, in handle_requests
    request = self.picamera2.process_requests()
  File "/home/pi/picamera2/picamera2/picamera2.py", line 600, in process_requests
    if function():
  File "/home/pi/picamera2/picamera2/picamera2.py", line 697, in <lambda>
    functions = [(lambda: self.switch_mode_(camera_config)),
  File "/home/pi/picamera2/picamera2/picamera2.py", line 676, in switch_mode_
    self.start_()
  File "/home/pi/picamera2/picamera2/picamera2.py", line 519, in start_
    raise RuntimeError("Camera did not start properly.")
RuntimeError: Camera did not start properly.
Aborted

JamesRayn, 2022-04-16:
In summery, I compared the log from different crashes. I see differences in the part "Unable to request 1 buffers: Cannot allocate memory" at the sample it is every time 1 buffer, at my script by doing only an switch_mode the output will say "Unable to request 4 buffers: Cannot allocate memory"
The next point who differs is between the line 600 and line 676 message, some times it reports instad the line 697 the following:
  File "/home/pi/picamera2/picamera2/picamera2.py", line 682, in `<lambda>`
    functions = [(lambda: self.switch_mode_(camera_config))]

davidplowman, 2022-04-19:
Hi, thanks for adding that information. It looks like a pretty clear case of being unable to allocate image buffers in CMA memory, though I have to confess I'm a little surprised to see this on a Pi 4. Before looking at anything else, can we just check whether you've changed anything in your /boot/config.txt file, or is it exactly as it was when you first installed Bullesye? Thanks!

davidplowman, 2022-04-19:
I have reproduced this now. I'm still thinking it's an interaction of camera image buffers and graphics memory that may be causing fragmentation or something like that, but it will need some investigation.

davidplowman, 2022-04-19:
I think I have a fix for this here: https://github.com/raspberrypi/picamera2/pull/68

JamesRayn, 2022-04-19:
Wonderful, looks good at the first view in normal operation, I will test it further soon. I still spotted crashes when I switch it very rapidly, but I have to check my skript and maybe, if there is a problem, I write a skript to replicate the case.
To answer the previous question anyway, I only modified the Video RAM setting to 256MB.

davidplowman, 2022-04-20:
Thanks. If you are able to recreate anything else re-producible, please do file a new issue for it.

You mentioned increasing "video RAM" - do you mean you're setting gpu_mem in your /boot/config.txt? We generally advise against that these days as not very many things use gpu_mem any more. Some legacy software may still use it, but unless you know you're doing that it's probably worth seeing if you can remove it and you'll get back about 200MB of memory.

JamesRayn, 2022-04-20:
Yes I did it in /boot/config.txt
The reason was the legacy camera system wich I don't use any more, if everything works as expected.

closed 2022-04-23 by JamesRayn