raspberrypi/picamera2 #69: Potential modification to build instructions in README.md

simonpickering, 2022-04-20:
In the commands following "Next we need some DRM/KMS bindings:"

I assume there's a missing "sudo ninja -C build install" at the end.

It might also be worth adding a note at the start to tell people not to try building on a RPi Zero W (as I did) as it hangs doing something swap related during this section of the build (it worked on a RPi 3 fwiw)

davidplowman, 2022-04-20:
> I assume there's a missing "sudo ninja -C build install" at the end.

I'm fairly (though not completely) sure that I've just cut'n'pasted these instructions and they worked, though I may be mistaken. Have you tried it? I'll pay attention to what happens whenever I next run through all these instructions.

> 
> It might also be worth adding a note at the start to tell people not to try building on a RPi Zero W (as I did) as it hangs doing something swap related during this section of the build (it worked on a RPi 3 fwiw)

Fair point. In truth, doing anything like this on a Pi Zero is going to be painful. We're working on getting proper pre-built packages together for all this stuff, so hopefully it will become moot before too long! But thanks anyway, I'll try and remember to slip something in when I'm next editing this file.

davidplowman, 2022-05-03:
I'm going to close this one now. The installation instructions have been updated now that apt packages including Python bindings are now available in our public repositories - so no more C++ compilation required!

closed 2022-05-03 by davidplowman

simonpickering, 2022-05-03:
Great, sorry I didn't get a chance to try again without having done the install to answer your question