raspberrypi/picamera2 #71: Worth adding custom EXIF to capture_file?

barryhunter, 2022-04-20:
I would like to write some additional EXIF tags to the captured .jpg file (specifically GEO - lat/long etc from a GPS, and possibly some other arbiary notes etc), rather than writing the jpeg, then having to add the data afterwards. 

Of course, could just grab a raw image (with  `make_image`) and then write the file, but as the library already has the `capture_file` method, wondering if would be interest in adding support for writing arbitrary EXIF data? Seems would be a common use case, and would avoid users having to duplicate the code in `CompletedRequest.save()`

Propose to extend the `save()` method, something like:
```
def save(self, name, filename, exif_dict={})
...
# add some extra tags to the user supplied data
exif_dict["0th"][piexif.ImageIFD.Make] = "Raspberry Pi"
...
exif_dict["Exif"][piexif.ExifIFD.ISOSpeedRatings] = int(total_gain * 100)
exif = piexif.dump(exif_dict)
```
and of course add to `capture_file()` (and `capture_file_`) so the data is passed though

Just unsure if would add before/after `signal_function`, suspect before, (because most users won't use `signal_function`) 
`def capture_file(self, filename, name="main", wait=True, exif_dict={}, signal_function=signal_event):
`

Then users would 
```
gps_ifd = {
        piexif.GPSIFD.GPSVersionID: (2, 0, 0, 0),
        #...
}
exif_dict = {"GPS": gps_ifd}
picam2.capture_file("test.jpg", exif_dict = exif)
```
(above code is just proposal, not tested yet!) 

I'd be willing to do this as a PR. Just checking if this the right direction first.

davidplowman, 2022-04-21:
Hi, thanks for suggesting this. Yes, I think it sounds like a good idea. The one thing I want to avoid is to have lots of exif-handling code of our own (like in _libcamera-apps_ where the whole exif thing is unbelievably painful), but in the Python world the situation is much better. I'd certainly be happy with something where you can optionally supply a dict of tags using `piexif`, and these just get passed though, more or less without touching anything.

So yes, please do submit a PR. Make sure to update the `switch_mode_and_capture_file` methods too. Also, you may have seen it, but I've tried to outline some guidance for contributions [here](https://github.com/raspberrypi/picamera2#contributing) (nothing unreasonable, I hope).

davidplowman, 2022-07-26:
Hi again, I'm going to close this one for the time being. If you'd like to submit a PR as discussed, please feel free to do so. Thanks!

closed 2022-07-26 by davidplowman