raspberrypi/picamera2 #116: capture_jpeg.py:  numpy.ndarray size change.. Expected 48, got 40

slowrunner, 2022-05-15:
Running capture_jpeg.py throws exception:
```
$ ./capture_jpeg.py 
Traceback (most recent call last):
  File "/home/pi/Pi4BE32/Examples/picamera2/./capture_jpeg.py", line 6, in <module>
    from picamera2 import Picamera2, Preview
  File "/home/pi/picamera2/picamera2/__init__.py", line 1, in <module>
    from .picamera2 import Picamera2, Preview
  File "/home/pi/picamera2/picamera2/picamera2.py", line 14, in <module>
    from picamera2.encoders import Encoder
  File "/home/pi/picamera2/picamera2/encoders/__init__.py", line 3, in <module>
    from .jpeg_encoder import JpegEncoder
  File "/home/pi/picamera2/picamera2/encoders/jpeg_encoder.py", line 1, in <module>
    import simplejpeg
  File "/usr/local/lib/python3.9/dist-packages/simplejpeg/__init__.py", line 1, in <module>
    from ._jpeg import encode_jpeg
  File "simplejpeg/_jpeg.pyx", line 1, in init simplejpeg._jpeg
ValueError: numpy.ndarray size changed, may indicate binary incompatibility. Expected 48 from C header, got 40 from PyObject
```

PiOS Bullseye 32-bit: Linux Pi4BE32 5.15.32-v7l+ #1538 SMP Thu Mar 31 19:39:41 BST 2022 armv7l GNU/Linux  
Python 3.9.2  
python3-libcamera/stable,now 0~git20220510+22656360-1 armhf [installed]  
v4l2-python3==0.3.1  

```
$ libcamera-jpeg --list
[0:04:16.510502904] [975]  INFO Camera camera_manager.cpp:293 libcamera v0.0.0+3544-22656360
[0:04:16.526498464] [976] ERROR CameraSensor camera_sensor.cpp:591 'ov5647 10-0036': Camera sensor does not support test pattern modes.
[0:04:16.543963437] [976]  INFO RPI raspberrypi.cpp:1356 Registered camera /base/soc/i2c0mux/i2c@1/ov5647@36 to Unicam device /dev/media3 and ISP device /dev/media0
Available cameras
-----------------
0 : ov5647 [2592x1944] (/base/soc/i2c0mux/i2c@1/ov5647@36)
    Modes: 'SGBRG10_CSI2P' : 640x480 1296x972 1920x1080 2592x1944 
```

davidplowman, 2022-05-16:
Yes, some users have reported a problem with numpy. It seems like some packages have moved on to a version that isn't in the default OS image. Obviously we'll look into that before we next make up new images. I believe the workarounds for now are:
```
sudo pip3 install numpy --upgrade
sudo apt install -y libatlas-base-dev
```

slowrunner, 2022-05-16:
Thank you both - that did the trick for me.

closed 2022-05-16 by slowrunner