raspberrypi/picamera2 #124: Dependency to PyQt

mgoutelle, 2022-05-20:
Do you plan to propose a headless version of picamera2, without the dependency on PyQt which is only necessary for previews (if I understand correctly)?

dinapappor, 2022-05-20:
I am wondering about this too. I have no need for GUI stuff (my current code is based on mmalobj from original picamera ( I have no need for GUI stuff) which works fine without running X11 and so on (with preview). 

Pulling in QT and GL stuff is too heavy for my usecase. I just want the bare minimum.

davidplowman, 2022-05-20:
Hi, thanks for raising this. To understand the problem, it's not that things don't work, it's just that more code is being installed than you need, is that right? We can look into that.

To answer the other question, PyQt is only required for preview windows when running under X-Windows. (When not running X-Windows, you can still use the DRM/KMS preview window if you want.)

mgoutelle, 2022-05-20:
> To understand the problem, it's not that things don't work, it's just that more code is being installed than you need, is that right? We can look into that.

That's exactly my concern. Ideally, one could choose to install this dependency or not, via the `pip install` options. And the absence of `PyQT` would need to be caught at import's time if one did not install the proper options but try to open a preview window.

davidplowman, 2022-06-09:
I think this is resolved now, but please open another ticket if there are still problems. Thanks!

closed 2022-06-09 by davidplowman