raspberrypi/picamera2 #130: [BUG] Unable to re-open/start camera once stop or close is called

apage224, 2022-05-20:
**Describe the bug**
I am unable to cleanly stop or close the camera connection. Once I do any combination of `stop` or `close`, I get an error on subsequent opens. I have to reboot in order to recover.

**To Reproduce**

Below is a slimmed down version based on the `capture_headless.py` example:

```python
from picamera2 import Picamera2

picam2 = Picamera2()
config = picam2.still_configuration()
picam2.configure(config)

picam2.start() 
picam2.capture_file("demo1.jpg")
picam2.stop()

picam2.start() # <- This will raise an exception
picam2.capture_file("demo2.jpg")
picam2.stop()
```

**Expected behaviour**
To cleanly stop and close camera connection.

**Console Output, Screenshots**

Below I've included the stack trace. If I try to restart Python and create new Picamera instance I get a similar error when trying to configure ( `picam2.configure(config)` ) getting error `ERROR RPI raspberrypi.cpp:873 Failed to set format on ISP Output1: -22`. It seems like comm. to the camera isn't working?

```bash

[0:06:31.111655590] [74]  INFO Camera camera_manager.cpp:293 libcamera v0.0.0+3544-22656360
[0:06:31.145453871] [92]  WARN CameraSensorProperties camera_sensor_properties.cpp:163 No static properties available for 'imx477'
[0:06:31.145495092] [92]  WARN CameraSensorProperties camera_sensor_properties.cpp:165 Please consider updating the camera sensor properties database
[0:06:31.145516166] [92] ERROR CameraSensor camera_sensor.cpp:591 'imx477 10-001a': Camera sensor does not support test pattern modes.
[0:06:31.172111369] [92]  INFO RPI raspberrypi.cpp:1356 Registered camera /base/soc/i2c0mux/i2c@1/imx477@1a to Unicam device /dev/media0 and ISP device /dev/media1
[0:06:31.176780565] [74]  INFO Camera camera.cpp:1029 configuring streams: (0) 4032x3040-BGR888
[0:06:31.177196113] [92]  INFO RPI raspberrypi.cpp:760 Sensor: /base/soc/i2c0mux/i2c@1/imx477@1a - Selected sensor format: 4056x3040-SBGGR12_1X12 - Selected unicam format: 4056x3040-pBCC
[0:06:33.696041744] [92] ERROR V4L2 v4l2_videodevice.cpp:1852 /dev/video15[70:cap]: Failed to start streaming: Invalid argument
2022-05-20T19:46:55.308Z | ERROR    | Camera did not start properly.
---------------------------------------------------------------------------
RuntimeError                              Traceback (most recent call last)
Input In [2], in <cell line: 9>()
      6 picam2.capture_file("demo1.jpg")
      7 picam2.stop()
----> 9 picam2.start()
     10 picam2.capture_file("demo2.jpg")
     11 picam2.stop()

File /usr/local/lib/python3.9/dist-packages/picamera2/picamera2.py:570, in Picamera2.start(self, event_loop)
    568 if event_loop and not self.have_event_loop:
    569     self.start_preview(Preview.NULL)
--> 570 self.start_()

File /usr/local/lib/python3.9/dist-packages/picamera2/picamera2.py:551, in Picamera2.start_(self)
    549 else:
    550     self.log.error("Camera did not start properly.")
--> 551     raise RuntimeError("Camera did not start properly.")

RuntimeError: Camera did not start properly.

```

**Hardware :**
Raspberry Pi 4 B w/ 8GB connected to Raspberry HQ Camera (IMX477).

davidplowman, 2022-05-21:
Hi, thanks for reporting this. I've tried to reproduce the problem on a clean Bullseye install (both 32 and 64 bit) but so far without success. Perhaps we can investigate a few more things:

- Are you using an up-to-date Bullseye? (Maybe report the output of `uname -a` and `vcgencmd version`.)
- Are there any changes in your /boot/config.txt from the default settings?
- Is there anything else significant running when the failures occur?
- If possible, it would be interesting to see the output of `cat /proc/meminfo` before you run the failing script.
- Does the problem occur immediately after boot? Or does it take a while to show up?
- When the problem occurs, could you look at the bottom of the output from `dmesg`. The error message does suggest a problem down in the kernel v4l2 drivers, so there may be some hints there. If you see anything strange, please post it here. (If you run `sudo dmesg -C` before running the script, that would clear out all the preceding stuff leaving it easier to understand!)
- In your script, if you replace `still_configuration` by `preview_configuration`, does that still fail? That would rule out (or not) big memory allocations.
- If you are able to (I realise this takes a bit of time and relies on having some spare SD cards around), could you try a fresh Bullseye image too?

I've certainly seen cameras fail to start when it runs out of CMA memory. This heap is prone to fragmentation under some circumstances, though normally I believe the error message does suggest this. It might be worth trying to increase the amount of CMA that you have. To do this go to your /boot/config.txt, find the line that says `dtoverlay=vc4-kms-v3d` and replace it by (exactly as given) `dtoverlay=vc4-kms-v3d,cma-384` (and you can check it worked after rebooting with `grep Cma /proc/meminfo`). But as I've said, I'm not quite convinced that's the story here.

Thanks!

davidplowman, 2022-08-17:
Hi, is there still anything we can help you with here? If there is please let us know, otherwise I'll consider closing this issue shortly.

davidplowman, 2022-09-05:
I'm going to close this issue for now. If something else comes up, or there's something more to look at, please feel free to open a new issue. Thanks!

closed 2022-09-05 by davidplowman

optimuspaul, 2022-09-27:
@apage224 can you share you DockerFIle?  I've been trying to get it installed in a container with no luck.

tmihajloski, 2023-10-03:
Hi,

I am experiencing this issue as well. I have a python script that streams to an RTSP server from the camera for 10s then closes the camera. When I try to run the script again I get the error:
`ERROR V4L2 v4l2_videodevice.cpp:1047 /dev/video13[18:out]: Unable to set format: Invalid argument`

I can resolve this error by rebooting the Pi.

Here is my script:

```
#!/usr/bin/python3

import socket
import time
from pprint import *
from picamera2 import Picamera2
from picamera2.encoders import H264Encoder
from picamera2.encoders import MJPEGEncoder
from picamera2.outputs import FfmpegOutput
from picamera2.outputs import FileOutput
from libcamera import controls

picam2 = Picamera2()

try:
    video_config = picam2.create_video_configuration({"size": (1920, 1080)})

    picam2.configure(video_config)
    picam2.set_controls({"AfMode": controls.AfModeEnum.Continuous})
    encoder = H264Encoder(bitrate=1000000, repeat=True, iperiod=15)
    #encoder = MJPEGEncoder()

    output = FfmpegOutput("-f rtsp -rtsp_transport tcp rtsp://localhost:8554/live.stream")

    print("start streaming")
    picam2.start_encoder(encoder=encoder, output=output)
    picam2.start()
    time.sleep(10)

finally:
    print("closing camera")
    picam2.stop()
    picam2.stop_encoder()
    picam2.close()
```

davidplowman, 2023-10-04:
Hi, I'm afraid I don't have any kind of RTSP server installed so I don't seem to be able to try your example. However, I've tried outputting to a simple udp stream instead, and I am able to restart the script without problems. Perhaps you could try some investigations:

* If you output to a udp stream (e.g. `FfmpegOutput("-f mpegts udp://127.0.0.1:1234/"`), does that fail in the same way?
* In the failing case, could you post the full console output of it 1. starting correctly and 2. failing to start.
* Are you sure that the first time you run the script, it's terminating normally?
* Does `libcamera-hello` start the camera successfully in this case, or not?
* When the script fails to restart, are there any Python processes still running? Is there anything there which, after killing with `kill -9`, enables the system to work?
* Are any kernel problems showing up in `dmesg`?

Thanks!

tmihajloski, 2023-10-04:
Thanks @davidplowman 

I used the MediaMTX server, which works out of the box (https://github.com/bluenviron/mediamtx/releases). 

- I tried using mpegts and it works without issues. However, I need RTSP for my application. 
- See console output below
- The script doesn't terminate normally, I believe it goes in the finally statement to clean up, because the stream to the RTSP server stops before the sleep timeout completes. I think part of the issue is the frame size, because reducing it to 1024x768 it doesn't fail immediately on the first run.
- libcamera-hello gives the same error
- I don't see any running python processes 
- I am attaching the dmesg log, with errors.

Here is the normal output
```
[0:01:50.661178358] [2230]  INFO Camera camera_manager.cpp:297 libcamera v0.0.5+83-bde9b04f
[0:01:50.783340397] [2231]  INFO RPI vc4.cpp:437 Registered camera /base/soc/i2c0mux/i2c@1/imx708@1a to Unicam device /dev/media1 and ISP device /dev/media2
[0:01:50.783436559] [2231]  INFO RPI pipeline_base.cpp:1101 Using configuration file '/usr/share/libcamera/pipeline/rpi/vc4/rpi_apps.yaml'
[0:01:50.794932359] [2230]  INFO Camera camera.cpp:1033 configuring streams: (0) 1920x1080-XBGR8888 (1) 2304x1296-SBGGR10_CSI2P
[0:01:50.795580257] [2231]  INFO RPI vc4.cpp:565 Sensor: /base/soc/i2c0mux/i2c@1/imx708@1a - Selected sensor format: 2304x1296-SBGGR10_1X10 - Selected unicam format: 2304x1296-pBAA
start streaming
closing camera
```

Here is the failed output

```
[0:01:50.661178358] [2230]  INFO Camera camera_manager.cpp:297 libcamera v0.0.5+83-bde9b04f
[0:01:50.783340397] [2231]  INFO RPI vc4.cpp:437 Registered camera /base/soc/i2c0mux/i2c@1/imx708@1a to Unicam device /dev/media1 and ISP device /dev/media2
[0:01:50.783436559] [2231]  INFO RPI pipeline_base.cpp:1101 Using configuration file '/usr/share/libcamera/pipeline/rpi/vc4/rpi_apps.yaml'
[0:01:50.794932359] [2230]  INFO Camera camera.cpp:1033 configuring streams: (0) 1920x1080-XBGR8888 (1) 2304x1296-SBGGR10_CSI2P
[0:01:50.795580257] [2231]  INFO RPI vc4.cpp:565 Sensor: /base/soc/i2c0mux/i2c@1/imx708@1a - Selected sensor format: 2304x1296-SBGGR10_1X10 - Selected unicam format: 2304x1296-pBAA
start streaming
closing camera
```

[dmesg.log](https://github.com/raspberrypi/picamera2/files/12803801/dmesg.log)

davidplowman, 2023-10-04:
I don't actually see any difference between the good/bad outputs, is that right? Though there's clearly trouble in the kernel log.

Can you say what is actually happening, is the FFmpeg process failing with this leaving things in a bad state? Does catching and ignoring the error make the problems go away?

davidplowman, 2023-10-04:
Thanks for updating the log. It seems like it must be the V4L2 kernel crash that stops the camera from re-opening - it's not that anyone is still holding on to the camera.

I'm struggling at the moment to see how that could happen. If FFmpeg dies and that thread crashes, I can see that buffers don't get recycled properly and _maybe_ V4L2 could get unhappy. Though a kernel crash seems worse than I would have expected. But if you catch and ignore the error - so the thread keeps running and the only difference is that it doesn't get sent out - how can it have a bad effect like that? Have you also fixed the code so that the stop() function won't try to do anything bad?

Out of interest, what is the error? Is it a broken pipe trying to write to ffmpeg.stdin? Are you able to tell if the kernel crash happens immediately, or only when you try to stop the system?

tmihajloski, 2023-10-04:
I printed the script events in the syslog as well. Attached is the syslog snapshot where the crash occurs [syslog.log](https://github.com/raspberrypi/picamera2/files/12807286/syslog.log).

It seems to happen after the encoder_stop() call. 

I added the finally block originally to work around this issue, thinking that if something fails it would get the chance to stop and close everything gracefully.

Below is the code that I used for testing.

```
#!/usr/bin/python3

import socket
import time
from pprint import *
from picamera2 import Picamera2
from picamera2.encoders import H264Encoder
from picamera2.encoders import MJPEGEncoder
from picamera2.outputs import FfmpegOutput
from picamera2.outputs import FileOutput
from libcamera import controls
import logging
import logging.handlers

logger = logging.getLogger("CameraLogger")
logger.setLevel(logging.DEBUG)

log_handler = logging.handlers.SysLogHandler(address= '/dev/log')

logger.addHandler(log_handler)

logger.debug('Starting rtsp_camera.py')
picam2 = Picamera2()

try:
    video_config = picam2.create_video_configuration({"size": (1920, 1080)})

    logger.debug('Configuring the camera')
    picam2.configure(video_config)
    picam2.set_controls({"AfMode": controls.AfModeEnum.Continuous})
    encoder = H264Encoder(bitrate=800000, repeat=True, iperiod=15)
    
    #encoder = MJPEGEncoder()

    output = FfmpegOutput("-report -f rtsp -rtsp_transport tcp rtsp://localhost:8554/live.stream")
    
    #output = FfmpegOutput("-f mpegts udp://127.0.0.1:1234/")
    
    print ("start streaming")
    logger.debug('Starting the encoder')
    picam2.start_encoder(encoder=encoder, output=output)
    logger.debug('Starting the camera')
    picam2.start()
    logger.debug('Waiting to stream')
    time.sleep(120)
    logger.debug('Waiting done')

finally:
    print("stopping camera")
    logger.debug('stopping the camera')    
    picam2.stop()

    print("stopping encoder")
    logger.debug('stopping the encoder')
    picam2.stop_encoder()    

    print("closing camera")
    logger.debug('closing the camera')
    picam2.close()

    
    print("camera closed")
    logger.debug('rtsp_camera.py done!')
```

davidplowman, 2023-10-05:
I'm still struggling a bit to understand how FFmpeg disappearing spontaneously can affect the rest of the code if you're catching and ignoring the error. When you do this, are you ensuring that the call to stop doesn't subsequently do anything, given that there's nothing to stop?

tvoverbeek, 2023-10-05:
@davidplowman Did you look at the ayalog extract?
There are messages about a crash in the video-driver:
````
Oct  4 14:31:22 raspi-zero kernel: [  207.816523] videobuf2_common: driver bug: stop_streaming operation is leaving buf 0000000005ae5823 in active state
Oct  4 14:31:22 raspi-zero kernel: [  207.816534] videobuf2_common: driver bug: stop_streaming operation is leaving buf 00000000eb3446cd in active state
Oct  4 14:31:22 raspi-zero kernel: [  207.816542] videobuf2_common: driver bug: stop_streaming operation is leaving buf 0000000099f8a9f8 in active state
Oct  4 14:31:22 raspi-zero kernel: [  207.816549] videobuf2_common: driver bug: stop_streaming operation is leaving buf 00000000fbbe929a in active state
```
The  first run already leavea some buffers in the active state.
The stack trace might also give some clues

davidplowman, 2023-10-05:
Hi @tvoverbeek, thanks for replying!

Yes, I'm altogether a bit confused by the whole thing. It's the encoder which is being left in an unhappy state when that thread that dequeues its buffers crashes. It complains that its V4L2 queue hasn't been drained when it is stopped, which is fair enough, but that this should lead to a kernel crash seems a bit beyond the pail.

Even so, wrapping the failure to write to FFmpeg (because it seems to have died spontaneously?) really ought to make the error go away. The buffers will all get dequeued in the normal way by that thread, and this is all known to work because it's what we do all the time.

So why is it crashing in this case? Is it something else that is now misbehaving? I don't know. Maybe inserting some code to make the FfmpegOutput crash after (for example) 100 frames would make the error reproducible so that it could be investigated.

Someone else recently had a very similar problem, but for them catching the exception did seem to work, so I was intending to add that change. Yet I don't really see why this case would be different. Though things are also undergoing a bit of a Pi 5 / Bookworm upheaval at the moment, which just compounds the fun...

tmihajloski, 2023-10-05:
Here is the ffmpeg log when It crashes. I have a 10s wait to stream, and I can see that ffmpeg stops streaming within a few seconds of starting. 
[ffmpeg-20231005-091753.log](https://github.com/raspberrypi/picamera2/files/12818819/ffmpeg-20231005-091753.log)