raspberrypi/picamera2 #132: [HOW-TO] Set multiple parameters to video_configuration?

Fredrum, 2022-05-23:
Hello!

I'm just starting to try and use the picamera2 python packages.
First things is I wanted to try and configure a 720p60 h264 recording and also save out the timestamps. 

Like this commandline does,
`libcamera-vid -t 10000 --width 1280 --height 720 --framerate 60 --intra 10 -b 40000000 -o test.h264 --save-pts timestamps.txt`

I can't find any examples that sets multiple parameters to the recording.

I just tried this and it doesn't error but ffprobe reports 30fps.

video_config = picam2.video_configuration({"size": (1280, 720), "FrameDurationLimits": (16666, 33333)})  # 1000000 / 60

Any help with this would be amazing or if there's a example code somewhere to look at?

Cheers!

kbingham, 2022-05-23:
Setting a FrameDurationLimit of 16666, 33333 (where those are min/max) is the same as saying the camera can run anywhere between 30 and 60 FPS. (33333 is 30 FPS).

Try setting the min and max to the same value to enforce 60 FPS  with (16666, 16666).

Fredrum, 2022-05-23:
This is a imx219 (PiCamera v2) with
RPi4b, Freshly flashed Raspberry OS 64bit.

All I do in the code after is,
```
picam2.configure(video_config)
encoder = H264Encoder(20000000)
picam2.start_recording(encoder, 'test.h264')
...
picam2.stop_recording()
```

davidplowman, 2022-05-23:
I think you're doing the right things, but the problem may be that you're writing an h.264 file. This is purely the video codec bitstream and contains no timing information at all. Different programs interpret it differently, I guess ffprobe says "30fps" if it doesn't find a reason to say otherwise.

Fredrum, 2022-05-23:
Ok cool so I'll look at the included timestamp example and see if that sorts it out! Cheers!

The other related things was, can I also do the equivalent of '--intra 10' ?
Is there a naming convention that should enable me to guess the Python control and config settings?

davidplowman, 2022-05-23:
> Ok cool so I'll look at the included timestamp example and see if that sorts it out! Cheers!

Yes, the timestamps should tell you exactly what is going on.

> The other related things was, can I also do the equivalent of '--intra 10' ? Is there a naming convention that should enable me to guess the Python control and config settings?

Documentation is still, I'm afraid, a work in progress. But it is Python code, so it's not too hard to look at the sources: https://github.com/raspberrypi/picamera2/blob/main/picamera2/encoders/h264_encoder.py#L6

Fredrum, 2022-05-23:
Oh yeah greak ok thanks!
I'm not used to Python programming yey.
Cheers!

closed 2022-05-23 by Fredrum