raspberrypi/picamera2 #137: [OTHER] vnc | camera legacy

Bra1nsen, 2022-05-24:
Unable view the desktop in VNC when I enable Legacy Camera in Pi 4 model...

Iam not the only one ..

https://www.reddit.com/r/raspberry_pi/comments/tw7ejh/unable_view_the_desktop_in_vnc_when_i_enable/

https://raspberrypi.stackexchange.com/questions/136649/vnc-not-able-to-show-desktop-when-legacy-camera-is-enabled

tvoverbeek, 2022-05-24:
Does not belong here since it has nothing to do with picamera2.

Anyway here an answer:
When you enable Legacy Camera support in bullseye it replaces the KMS driver with the previously used
fake KMS driver (`dtoverlay=vc4-fkms-3d` instead of `dtoverlay=vc4-kms-3d` in /boot/config.txt).
If you are running headless then you get the cannot connect ....
When using the fkms driver you have to use the legacy hdmi options in /boot/config.txt:
hdmi_froce_hotplug, hdmi_group, hdmi_mode to force a resolution for vnc.
Check you have the latest realvnc server (6.9.1) installed.
WIth the legacy hdmi options I get a proper desktop on a headless pi4 with legacy camera support enabled.

Bra1nsen, 2022-05-24:
thanks for reply,

so i activate legacy and change dtoverlay in /boot/config.txt - thats it?

tvoverbeek, 2022-05-24:
The dtoverlay change is done by raspi-config when you enable Legacy Camera support.
You only have to set the proper hdmi options in /boot/config.txt. They are commented out in the default /boot/config.txt

Bra1nsen, 2022-05-24:
great it worked! love you

closed 2022-05-24 by Bra1nsen