raspberrypi/picamera2 #138: [OTHER] Resolution 4032x3040

Bra1nsen, 2022-05-25:
#!/usr/bin/python3

# Capture a DNG and a JPEG made from the same raw data.

from picamera2 import Picamera2, Preview
import time

picam2 = Picamera2()
picam2.start_preview(Preview.QTGL)

preview_config = picam2.preview_configuration()
capture_config = picam2.still_configuration(raw={}, display=None)
picam2.configure(preview_config)

picam2.start()
time.sleep(2)

r = picam2.switch_mode_capture_request_and_stop(capture_config)
r.save(name="main", filename="full.jpg")
r.save_dng(filename="full.dng")

1. Is it possible to capture images without using preview?
2. Why do I get 4032x3040, I should get 4056x3040

Wi-Fi is currently blocked by rfkill.
Use raspi-config to set the country before use.

[0:12:20.526728361] [2812]  INFO Camera camera_manager.cpp:293 libcamera v0.0.0+3544-22656360
[0:12:20.543309989] [2813]  WARN CameraSensorProperties camera_sensor_properties.cpp:163 No static properties available for 'imx477'
[0:12:20.543356229] [2813]  WARN CameraSensorProperties camera_sensor_properties.cpp:165 Please consider updating the camera sensor properties database
[0:12:20.543386377] [2813] ERROR CameraSensor camera_sensor.cpp:591 'imx477 10-001a': Camera sensor does not support test pattern modes.
[0:12:20.559965264] [2813]  INFO RPI raspberrypi.cpp:1356 Registered camera /base/soc/i2c0mux/i2c@1/imx477@1a to Unicam device /dev/media0 and ISP device /dev/media1
[0:12:22.053185534] [2812]  INFO Camera camera.cpp:1029 configuring streams: (0) 640x480-XBGR8888
[0:12:22.060285050] [2813]  INFO RPI raspberrypi.cpp:760 Sensor: /base/soc/i2c0mux/i2c@1/imx477@1a - Selected sensor format: 2028x1520-SBGGR12_1X12 - Selected unicam format: 2028x1520-pBCC
qt.qpa.xcb: QXcbConnection: XCB error: 148 (Unknown), sequence: 192, resource id: 0, major code: 140 (Unknown), minor code: 20
[0:12:25.042919247] [2816]  INFO Camera camera.cpp:1029 configuring streams: (0) 4032x3040-BGR888 (1) 4056x3040-SBGGR12_CSI2P
[0:12:25.045845856] [2813]  INFO RPI raspberrypi.cpp:760 Sensor: /base/soc/i2c0mux/i2c@1/imx477@1a - Selected sensor format: 4056x3040-SBGGR12_1X12 - Selected unicam format: 4056x3040-pBCC

3. I use Ethernet and turned off wifi, what is this warning good for?
4. no static proberties available? sensor does not support test patters?

davidplowman, 2022-05-25:
> **1. Is it possible to capture images without using preview? 2. Why do I get 4032x3040, I should get 4056x3040**

Yes, as shown below. We often wait a little while after starting the camera for the first time because things like AGC/AEC and AWB need a moment to converge. But you can ask for a capture immediately.

It recommends 4032x3040 because 4032 is aligned with what the hardware will produce and can be used more efficiently. You can ask for 4056x3040, but the hardware will have to add padding onto the ends of the rows (it must produce 
a multiple of 64 bytes). Many Python modules don't really understand this, so an extra copy of all the pixel data may be required in order to use it. The example below also shows how to force 4056x3040.
```
from picamera2 import Picamera2

picam2 = Picamera2()
config = picam2.still_configuration()
config["main"]["size"] = (4056, 3040)
picam2.configure(config)
picam2.start()
picam2.capture_file("test.jpg")
```
> ```
> <snip>
> ```
> **3. I use Ethernet and turned off wifi, what is this warning good for? 4. no static proberties available? sensor does not support test patters?**

The Ethernet warning has, so far as I know, nothing to do with the camera system and I can't tell you anything about it. It might be worth asking on a more general Raspberry Pi forum.

Those other error messages are coming from libcamera which is a bit verbose at times. We'll talk to them and see if we can improve it, but this can take some time. They can be safely ignored.

Bra1nsen, 2022-05-25:
ok thanks mate!

closed 2022-05-25 by Bra1nsen