raspberrypi/picamera2 #145: [HOW-TO] ExposureTime / ShutterSpeed

Bra1nsen, 2022-05-27:
How set minimum ExposureTime / maximum ShutterSpeed.

**Dear Developers**, I working on a SkyCam for SolarPowerPredictions. I created some examples..

`libcamera-jpeg -o shutterspeedxy --shutter xy`

[image]

Ironically there comes a point where the exposure increases again instead of continuing to decrease.

However I just would like to reduce the exposure time further? How can I set the maximum possible, do i need to add more settings?

Thank you!

davidplowman, 2022-05-27:
Hi, I notice you're using libcamera-apps, so [this](https://github.com/raspberrypi/libcamera-apps/issues) might be a more appropriate place to file a question.

However, I do notice that it seems like you're changing the shutter but not fixing the gain. I'm guessing this might not be what you intend, so try adding `--gain 1` to the commands. I don't know if this would fix what you're seeing but it's probably the first thing to try. Please post back on the libcamera-apps page if you'd like more advice after that.

Thanks!

Bra1nsen, 2022-05-27:
Why should gain change with different shutter speeds? Just tested it, it makes no difference...

davidplowman, 2022-05-27:
Because if you fix the shutter speed, it will try to use the gain to achieve the same brightness target values instead.

Bra1nsen, 2022-05-27:
Okay make sense david, anythingelse I could try?

davidplowman, 2022-05-27:
I see you've posted on libcamera-apps now, so thanks for doing that. I've replied there, so will close this thread. Thanks!

closed 2022-05-27 by davidplowman

davidplowman, 2022-06-02:
There are a couple of problems here. Firstly it looks like you might be using the old PiCamera interface? The Picamera2 interface uses a set_controls method to pass particular camera controls, e.g.
```
camera.set_controls({"ExposureTime": 100000, "AnalogueGain": 1})
```
The next thing is that cameras have quite a deep pipeline, especially for exposure and analogue gain (as even the sensor takes several frames to deal with them), so it will be quite a few frames before those controls have the effect you want. Though you can change them frame-by-frame, so the long-ish delay is only for the first change to happen. (The other alternative is to stop the camera, request the values and then restart it. You're then guaranteed that the very first frame will have to values you wanted, but the whole stop/start procedure is a bit slow.)

I've wondered about providing a function which says "capture frames at these different exposures/gains". It would take a while for the first one to arrive, but then they'd come in pretty much "back to back", (Though even then care has to be taken because nothing guarantees that the system might not drop one of them, so you'd have to sort of keep requesting them until you've got them all). So maybe I'll look into that.

One more small fly in the ointment is that I think nothing actually tells you the max/min possible exposures, so asking for something smaller than is possible would end up getting stuck waiting for something that can't happen. We'd probably need some changes in libcamera to address that, so that might be slightly trickier, or at least take longer to change.

Bra1nsen, 2022-06-02:
Is there any documentation that I can't find or how could I have known about set_controls?

Ok I'll see how long it takes with start stop or try the frame rate.

You definitely have your hands full, but actually such a function would be great and actually not that complicated.

Regarding your last comment, I think it would make sense if there was a fixed value for the minimum shutter speed, and any value selected below that would automatically assume this.

were you able to find out the minimum exposure time for the imx477?

davidplowman, 2022-06-02:
You need to set the "FrameDurationLimits". It takes a pair of values being the min and max frame durations in microseconds. You can set them to be the same if you want.

Bra1nsen, 2022-06-02:
**thanks you.** where can I find a list with information about the all possible settings?

davidplowman, 2022-06-02:
You can look at `camera.camera_controls` to see what's available. Formal documentation beyond the GitHub readme is something I'm only just starting on. The camera controls are supposed to show you the min and max values, though I note for exposure time and frame durations it only seems to show placeholder values.

davidplowman, 2022-06-02:
I think just ask for the `camera_controls` property of the `Picamera2` object. (sorry, not at a Pi right now).