raspberrypi/picamera2 #147: RuntimeError: Control 'DigitalGain' not found

cpixip, 2022-05-29:
When trying to set analog-, digitalgain and exposure with the following script:

```
#!/usr/bin/python3

from picamera2 import Picamera2, Preview
import time

picam2 = Picamera2()
picam2.start_preview(Preview.QT)

preview_config = picam2.preview_configuration()
picam2.configure(preview_config)

picam2.start()
time.sleep(1)
metadata = picam2.capture_metadata()
print(metadata)

time.sleep(1)
controls = {}
controls['AnalogueGain'] = 1.0
controls['DigitalGain']  = 2.0
controls['ExposureTime'] = 3000

print(controls)
picam2.set_controls(controls)
time.sleep(1)
metadata = picam2.capture_metadata()
print(metadata)
time.sleep(5)
```
I get the error "RuntimeError: Control 'DigitalGain' not found". When commenting out line 20, `controls['DigitalGain']  = 1.0`, everthing works like expected. I am using a HQ camera attached to a PI 3; setting both gains was no problem in the original picamera-library. The opencv_mertens_merge.py uses only AnalogueGain - is setting DigitalGain something libcamera does not allow for?

davidplowman, 2022-05-30:
Hi, thank you for your question. You can't set the digital gain, you can only set the analogue gain which is actually interpreted as a combined gain. The gain that you pass is used entirely as analogue gain within the sensor, and this normally gives a better result in terms of image quality than applying it later as digital gain. Once the analogue gain reaches whatever limit the sensor has, then the extra gain is applied as digital gain. Our documentation talks a little more about this subject [here](https://www.raspberrypi.com/documentation/accessories/camera.html#exposure-control).

cpixip, 2022-05-30:
thank you for this information - that was very helpful. 

Indeed, as you mentioned, analog and digital gains are acting differently on an image. 

Anyway, in my tests, I noticed that when setting only analog gain and exposure time (as in the exposure_fixed.py example), the total gain nevertheless varied between 1.000 and 1.016 with different captures/exposures. The reason was that while the analog gain was kept at 1.0, the digital gain varied.

Well, while reading the documentation you referenced, I noticed that in the libcamera-application suite's "common command line section" there are two options referenced: %ag = analog gain and %dg = digital gain. So it seems to be possible with the libcamera-apps to set these two gains independently? 

Some background: I have a film scanner project for scanning color-reversal movie stock. Currently it uses the old picamera-library. Due to the high dynamic range of such color-reversal material, you have to take several different exposures of a single frame and merge them into a single HDR image. For this to function properly, you need to keep analog as well as digital gain as constant as possible during all exposures.  Only the exposure time is varied between different captures. Between 5 and 6 exposures of a single movie frame are used for a HDR capture. That's why I am interested in being able to set the digital gain as well (more precisely: keep it fixed to the default value). Is there maybe a way through the camera tuning file to stop the ACG from interfering/changing the digital gain?

davidplowman, 2022-05-30:
> Anyway, in my tests, I noticed that when setting only analog gain and exposure time (as in the exposure_fixed.py example), the total gain nevertheless varied between 1.000 and 1.016 with different captures/exposures. The reason was that while the analog gain was kept at 1.0, the digital gain varied (presumably via the AGC?)

When the AGC/AEC is running, it calculates the total exposure time it thinks it wants, and divides this up into actual exposure and gain, according to the exposure profile. It asks the camera for that shutter time, but in practice that gets quantised because the sensor can only do certain fixed shutter times (normally multiples of the time it takes to read out a single row of pixels).

Anyway, the "difference" (ratio of the exposure time wanted divided by what it got) gets taken up by the analogue gain, and the AGC asks for that. Now, this generally ends up being quantised too because the analogue gain is only available in certain steps (even if relatively fine). Here too, it calculates the ratio of the analogue gain it wanted compared to the gain it got, and - you've probably guessed by now - this becomes the digital gain. The analogue gain steps are generally quite fine so the digital gain will stay very close to 1.

> Well, while reading the documentation you referenced, I noticed that in the libcamera-application suite's "common command line section" there are two options referenced: %ag = analog gain and %dg = [...]

In libcamera-apps you can print out the analogue gain and digitial gain separately. Indeed in Picamera2 you can get the gains separately from the metadata But you can't specify a digital gain value in libcamera-apps etiher, it's the same AGC procedure.

> ... For this to function properly, you need to keep analog as well as digital gain as constant as possible during all exposures. Only the exposure time is varied between different captures. Between [...]

You could fix the digital gains in the exposure profiles in the tuning, but that sounds over-complicated. If you really want to keep analogue and digital gain constant, you might do best by simply fixing the analogue gain (which is of course the combined gain) e.g.

```
...
picam2.start()
time.sleep(1)  # give the AGC/AEC a moment to settle fully
metadata = picam2.capture_metadata()
combined_gain = metadata["AnalogueGain"] * metadata["DigitalGain"]
picam2.set_controls({"AnalogueGain": combined_gain})
...
# If you want to let the gains vary again:
picam2.set_controls({"AnalogueGain": 0})
```
There are some things to be aware of when changing the exposure too. If you do it while the camera is running then it can take quite a few frames to take effect. If you do it while the camera is stopped, you're guaranteed that the very first frame will have the values you want (I assume you've seem the Mertens merge example). Stopping and restarting the camera is not immediate of course, but probably no slower than waiting for lots of intermediate frames before you get the one you want.

cpixip, 2022-05-30:
Thank you for the very detailed explanation. That clarifies things a lot for me!

So I shouldn't be too concerned about the digital gain variations, as these are just tiny corrections to the available discrete exposure times. 

I will try to move my software from the old picamera to the newer picamera2 Software and will report the results to the Kinograph forum, where quite a few other people are using the Raspberry Pi HQ camera with Python for film scanning projects.

Again, thanks a lot!

davidplowman, 2022-05-31:
I'll close this issue now, but please do post back if there are further questions. I'm very keen to help groups of users move on and use Picamera2!

closed 2022-05-31 by davidplowman

snaggas, 2023-12-04:
Hi David,
I just noticed something that may be of interest to you. When using gains over the sensor limit of 16, let's say 32, the `libcamera-vid` call prints out an AnalogueGain of around 16 and a DigitalGain of around 2, in line with your explanation below. When doing the equivalent with `picamera2` and printing the gains from metadata, the DigitalGain has not changed from 1.0. The `picamera2` video does indeed look also look less bright, at least to my eye.
Regards,
Sam

> Hi, thank you for your question. You can't set the digital gain, you can only set the analogue gain which is actually interpreted as a combined gain. The gain that you pass is used entirely as [...]

davidplowman, 2023-12-04:
Hi, thanks for passing this on. I've just tried this for myself and as far as I can tell it behaves as expected. Could you perhaps include the most minimal script that you can that demonstrates the problem so that I can try to reproduce it. Thanks.

For reference, here's what I ran:
```
from picamera2 import Picamera2
picam2 = Picamera2()
picam2.start()
picam2.set_controls({'AnalogueGain': 32})
```
after which `picam2.capture_metadata()` reports analogue gain of roughly 22.26 and digital gain of 1.44x. I'm using a Pi 4 with an imx477 (HQ cam) and up-to-date software (Picamera2 0.3.16).

snaggas, 2023-12-04:
Thanks. This script reproduces it (AnalogueGain ~ 15.7, DigitalGain = 1.0) but it seems to be fixed by setting the gain after `picam2.start()`, not before (sorry if that was obvious):

```
from picamera2 import Picamera2
import time
import libcamera

picam2 = Picamera2()
picam2.configure(picam2.create_video_configuration(transform=libcamera.Transform(hflip=1, vflip=1)))
picam2.set_controls({'AnalogueGain': 32})
picam2.start()

time.sleep(1)

metadata = picam2.capture_metadata()
print(metadata["AnalogueGain"], metadata["DigitalGain"])
```
Note: I'm using the GS camera with a Pi 3B.