raspberrypi/picamera2 #151: [BUG] Current Distribution does not match Github

ktorimaru, 2022-05-31:
I recently installed picamera2 on my Pi and I am trying to learn the package by using the examples.

I discovered that the post_callback requests were not firing. I reviewed the code on my Mac and could not find the error. So I decided to try and debug on my Pi. I opened picamera2.py on the Pi only to discover post_callback does not exist instead the variable is request_callback.

My plan is to uninstall the distribution and clone the github version to the Pi but thought you would want to know.

davidplowman, 2022-06-01:
Yes, I'm sorry that the GitHub code is moving on but we haven't updated the installation repositories yet. Hopefully we'll be doing that soon. In the meantime you could either clone the version with tag v0.2.1, or install the bleeding-edge version with `pip3 install git+https://github.com/raspberrypi/picamera2.git`.

davidplowman, 2022-06-06:
Yes, I think that would be better. Once we've got the next release out of the door, we could move to doing development on a "next" branch, so I'll propose that.