raspberrypi/picamera2 #154: [BUG] Cannot allocate memory after capture_request() and release()

gkyle, 2022-06-02:
**Describe the bug**
Repeated calls to capture_request() prompt this V4L2 error occurs:

ERROR V4L2 v4l2_videodevice.cpp:1211 /dev/video14[16:cap]: Unable to request 1 buffers: Cannot allocate memory
ERROR RPI raspberrypi.cpp:1008 Failed to allocate buffers
ERROR    | Camera did not start properly.

**To Reproduce**
Here is a short test case. If I make repeated calls to switch_mode_and_capture_image(), I can capture stills indefinitely. When I use switch_mode() and capture_request(), the V4L2 error occurs after 4 or 5 iterations.

from picamera2 import Picamera2, Preview
import time

camera = Picamera2()
camera.start_preview(Preview.DRM)
preview_config = camera.preview_configuration()
capture_config = camera.still_configuration(raw={})
camera.configure(preview_config)

def capture_img():
    pilImage = camera.switch_mode_and_capture_image(capture_config)
    print(pilImage)

def capture_req():
    camera.switch_mode(capture_config)
    request = camera.capture_request()
    pilImage = request.make_image("main")
    print(pilImage)
    request.release()

camera.start()

while True:
    capture_req()
    time.sleep(2)

camera.close()

**Expected behaviour**
Memory should be freed by CompletedRequest.release().

**Hardware :**
Compute Module 4 w/ 256MB allocated to GPU. PiCam HQ

davidplowman, 2022-06-03:
Hi, can you attach the (non-comment) lines of your /boot/config.txt, please? Thanks! Also, is this a reasonable fresh Bullseye installation or do you have other things running? Thanks!

davidplowman, 2022-06-03:
Oh, and can you say how much memory your CM4 has? Maybe post the output of `cat /proc/meminfo`. Ta!

davidplowman, 2022-06-03:
I've taken a quick look and I think the basic problem is fragmentation of the CMA heap. The camera system is always somewhat at risk of this when we're displaying camera images and therefore holding on to them whilst trying to allocate new camera buffers when we switch camera mode.

Mostly this works fine but the use case here, where the display is trying to hang on to a 12MP buffer, and then we keep allocating ever more stuff after this, it simply runs out of unfragmented memory. I'm not sure that there are any easy fixes for this, though there are numerous workarounds:

- Simply don't display the 12MP buffers. You may have noticed that the DRM/KMS display system doesn't seem to cope with images that big in any case, so there doesn't seem to be any point in trying. To give this a go, change your capture_config to look like this: `capture_config = camera.still_configuration(raw={}, display=None)`

- If you really do want to display the capture images, it would work better to generate a downscaled version and display that. The Pi's camera system can do that automatically for you. Try `capture_config = camera.still_configuration(raw={}, lores={"size": (1024, 768)}, display="lores")`

- In your example you keep switching to the same camera mode. Perhaps you're only doing this as a test? But in real life there's no point in it as it forces all the buffer allocations to get torn down and recreated, giving fragmentation a change to creep in every time. So the correct procedure would be to switch to the capture mode just once, and then to leave it there.

Hope that helps!

davidplowman, 2022-06-03:
(Unfortunately some of what I wrote above may only apply to the development version of Picamera2, not to the currently released version. We'll try and get an update pushed through in the coming days. Sorry about that.)

gkyle, 2022-06-06:
Addressing questions about my configuration: Yes. Recent Bullseye. I am using the **dev** version of PiCamera. Contents of /boot/config.txt:

```
dtparam=i2c_arm=on
dtoverlay=i2c-gpio,bus=3,i2c_gpio_delay_us=1
gpio=23,24=pu
dtoverlay=imx477,cam0
display_auto_detect=1
dtoverlay=vc4-kms-v3d
disable_overscan=1
[cm4]
otg_mode=1
[all]
hdmi_force_hotplug=1
hdmi_group=2
hdmi_mode=82
hdmi_drive=2
gpu_mem=256
```

gkyle, 2022-06-06:
Thanks. I can confirm that adding display=None to capture_config addresses the issue. I seem to be able to switch/capture indefinitely this way. I think we can resolve this issue.

One quick question about the intended use of the API. I think it's a common use case for an application to capture an image then 1) display the image (down-scaled) to a user, then 2) write the image to disk, then perhaps 3) write an alternate stream to disk (eg. jpeg + raw). The behavior of the switch_mode_and_capture_X methods each provide for one of those steps. These methods make use of helpful methods in CompletedRequest for converting a stream to a type of image and writing an image format to disk. For an application to perform 2+ steps of this use case, it is desirable to interact with the CompletedRequest directly. Is that what the developers intend?

Incidentally, those helpful methods (make_image, save, save_dng) in CompletedRequest don't require the full state of the request to operate. They could take in a buffer and metadata instead and be decoupled from CompletedRequest. I'm happy to show a small refactor that would do this, so API users don't have to interact with CompletedRequest directly, if that is of interest. Thanks!

davidplowman, 2022-06-06:
Thanks for the reply. I think I'm probably going to change the `still_configuration` function to supply `display=None` by default. It is, as we've discovered, a bit problematic and there are usually better ways to display such images (by creating the downscaled version, for example).

One small point about your config.txt, though it's not making any real difference here, but I would consider removing the `gpu_mem=256` like, unless you had some particular reason to add it. The new camera stack doesn't use gpu mem, so you are probably consigning about 200MB to the dustbin where it will never get used. I would take the line out and the system will decide.

As regards adding some new helper methods, could give some examples how they would be used? Generally speaking I'm expecting that some methods (e.g. switch_mode_and_capture_file) are more for the convenience of very simple use cases, and perhaps folks coming from the old PiCamera. Increasingly I would expect capturing requests to be important, because you get the corresponding images from all the streams, as well as all the metadata for exactly those images. I'm certainly happy to add new useful APIs - just want to understand exactly what we want to do with them before anyone does loads of work! Thanks!

gkyle, 2022-06-14:
Thanks for the details. It sounds like the API surface is working as intended. I wrote up a quick draft of the change I mentioned, in case it's of interest, here:

https://github.com/raspberrypi/picamera2/pull/171

closed 2022-06-14 by gkyle