raspberrypi/picamera2 #156: [BUG]

cpixip, 2022-06-03:
Installed picamera2-0.2.1; when running "app_capture_request.py", I got the following error

```
>>> %Run app_capture_request.py
Traceback (most recent call last):
  File "/home/pi/picamera2/examples/app_capture_request.py", line 10, in <module>
    from picamera2.previews import QPicamera2
ImportError: cannot import name 'QPicamera2' from 'picamera2.previews' (/usr/local/lib/python3.9/dist-packages/picamera2/previews/__init__.py)
```
It works if line 10 is changed from 
`picamera2.previews import QPicamera2`
to 
`picamera2.previews.qt import QPicamera2`

davidplowman, 2022-06-06:
Thanks for catching this !  https://github.com/raspberrypi/picamera2/pull/159

davidplowman, 2022-06-07:
We've merged that so I'll close this. An updated release to PyPI should be out shortly.

closed 2022-06-07 by davidplowman