raspberrypi/picamera2 #163: [HOW-TO] Change MJPEGserver.py to H.264

gallitim, 2022-06-07:
So I am relatively new to this streaming business, but I've been having struggles with fps and latency when using the mjpegserver.py file. Is there an easy way to increase to about 30 fps consistently or to change the format to use the h264 encoder? I'm basically just looking for any way to have a smoother stream over http.

davidplowman, 2022-06-07:
Hi, are you talking about https://github.com/raspberrypi/picamera2/blob/main/examples/mjpeg_server.py?

First of all it might be worth saying just what kind of problems you're having with performance/latency. On a Pi 4 I wouldn't expect any performance problems unless you've increased the resolution a lot, but perhaps you're using a lower powered device? By latency I assume you're referring to the receiving end - again, what kind of device are you using there and are you just viewing it in a browser?

Converting to H.264 is probably difficult in a web page because I don't think you can just stream frames as part of an HTTP message in the same way, but perhaps other more knowledgeable folks can help you there. Often people might simply stream vanilla H.264 or some other encapsulated format to a socket and then play them with something like vlc, can you say more about what your application is?

gallitim, 2022-06-08:
Yeah, I am running it on a pi4 with 4gb of ram, I'm aiming for a 720p stream at 30fps with as little latecny as possible, so I altered the code slightly to meet that higher resolution. I am using a default power supply but getting a very choppy stream when I connect to a browser locally through the pi. Http streaming would work best for me, and I technicnally don't have to stream in any specific format. Are any of these issues possibly affecting the performance? Is the resolution to high or is it simply an issue with my network?

Also just to make sure, the default framerate for this program is 30 fps since the frame duration limit is set to 33333 right? If this is true, are there any other paremeters in video_configuration that I can use?

Any help would be greatly appreciated.

davidplowman, 2022-06-08:
I have a Pi4 4GB too, and I've changed those two occurrences of 640 and 480 to 1280 and 720. I'm viewing it on my laptop having pointed my browser at socket 8000 on the Pi. I wasn't quite clear whether you were viewing it on the server Pi or a different Pi or something else?

It seems to be running really well for me, the stream is steady, framerate looks fine, latency maybe between 100 and 200ms. On the Pi it's doing software JPEG encode so the cores are fairly busy (according to htop), but nothing looks close to maxed out (looking at the 4 bars at the top). What does yours look like?

If I run the client on a (different) Pi 4 it's pretty good, though I would say not quite as good as the laptop. The Pi isn't maxed out either, I guess the browser is doing a less good job of playing the stream back.

gallitim, 2022-06-08:
I was originally viewing it on the pi running the server, when I view it from my laptop, the stream is usually a bit smoother but often freezes up while im running my hand over the camera sensor. My htop looks fine as well, so maybe its a camera hardware issue. Is your pi connected to your network through ethernet?

davidplowman, 2022-06-09:
Mine is connected with wired ethernet, it might be that works better. There is also other MJPEG streaming software out there that people have used, I don't know anything about it but there'd likely be some folks on the Camera Board forum who would know.

davidplowman, 2022-08-22:
Hi, are there any updates on this issue or have you been able to resolve of work around them? Please let us know if there's anything else we can look at here, otherwise I'll consider closing the issue in a few weeks if there's nothing to report. Thanks!

davidplowman, 2022-09-06:
I think I'm going to close this one for the time being. But if anything else comes up that we can look into, please feel free to open a new issue. Thanks!

closed 2022-09-06 by davidplowman