raspberrypi/picamera2 #166: [BUG] Camera __init__ sequence did not complete

911GT2, 2022-06-09:
After installing picamera2 as suggested I also had to install libcamera from source (and thus kmsxx and all the dependencies).
I now have a working libcamera installation and all the cli commands provide images.

After trying the following code 
`
from picamera2 import Picamera2, Preview

ppcam = Picamera2()
ppcam.start()
`

I have this

> [0:07:42.544557834] [1455]  INFO IPAManager ipa_manager.cpp:138 libcamera is not installed. Adding '/home/pi/libcamera/build/src/ipa' to the IPA search path
[0:07:42.545470529] [1455]  INFO Camera camera_manager.cpp:293 libcamera v0.0.0+3630-d0b310b4
[0:07:42.554329995] [1459]  INFO IPAProxy ipa_proxy.cpp:130 libcamera is not installed. Loading IPA configuration from '/home/pi/libcamera/src/ipa/raspberrypi/data'
[0:07:42.563849801] [1459]  INFO RPI raspberrypi.cpp:1356 Registered camera /base/soc/i2c0mux/i2c@1/ov5647@36 to Unicam device /dev/media2 and ISP device /dev/media0
2022-06-09T10:41:06.356Z | ERROR    | Camera __init__ sequence did not complete.
Traceback (most recent call last):
  File "/home/pi/.local/lib/python3.9/site-packages/picamera2/picamera2.py", line 77, in __init__
    self.open_camera()
  File "/home/pi/.local/lib/python3.9/site-packages/picamera2/picamera2.py", line 164, in open_camera
    if self.initialize_camera():
  File "/home/pi/.local/lib/python3.9/site-packages/picamera2/picamera2.py", line 148, in initialize_camera
    self.sensor_resolution = self.camera.properties["PixelArraySize"]
KeyError: 'PixelArraySize'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/pi/cam.py", line 3, in <module>
    ppcam = Picamera2()
  File "/home/pi/.local/lib/python3.9/site-packages/picamera2/picamera2.py", line 81, in __init__
    raise RuntimeError("Camera __init__ sequence did not complete.")
RuntimeError: Camera __init__ sequence did not complete.

I just don't know what to do, if anyone has any ideas I would be grateful ! Thanks

davidplowman, 2022-06-09:
Hi, where did you install libcamera from? Picamera2 is currently not up-to-date with the latest mainline libcamera, so you have to use the picamera2 branch from https://github.com/raspberrypi/libcamera. Obviously we'll be endeavouring to catch up with libcamera shortly.

911GT2, 2022-06-09:
> Hi, where did you install libcamera from? Picamera2 is currently not up-to-date with the latest mainline libcamera, so you have to use the picamera2 branch from https://github.com/raspberrypi/libcamera. Obviously we'll be endeavouring to catch up with libcamera shortly.

Thanks for the fast reply, I follow [this guide](https://www.raspberrypi.com/documentation/accessories/camera.html#building-libcamera-and-libcamera-apps) from "Building libcamera" to just before "Building libepoxy". 

I did all this because I have to use the 64bit image to run my tensorflow model, should i restart from a clean image and use only the picamera branch to install both?

davidplowman, 2022-06-09:
Yes, I'm afraid the README for this repository mentions that you have to change a couple of things in the standard instructions. We are starting to work on documentation, but things are all a bit bleeding edge at the moment. Maybe we should update the standard instructions, I guess we haven't totally decided whether to curate our own branch of libcamera permanently (we don't really want to).

Is there any reason you can't do the [default install](https://github.com/raspberrypi/picamera2#installation) (again from this repository's README)? I believe everything works on 64-bit.

911GT2, 2022-06-09:
Just reinstalled a fresh copy of 2022-04-04-raspios-bullseye-arm64-lite and yep it works with the following test code even though I have some "error" in the print like ( [7699] ERROR CameraSensor camera_sensor.cpp:591 'ov5647 10-0036': Camera sensor does not support test pattern modes. )

```
from picamera2 import Picamera2, Preview
import time

picam2 = Picamera2()

preview_config = picam2.preview_configuration(main={"size": (800, 600)})
picam2.configure(preview_config)

#picam2.start_preview(Preview.QTGL)

picam2.start()
time.sleep(2)

metadata = picam2.capture_file("test.jpg")
print(metadata)

picam2.close()
```

Anyway I didn't realise how early this all is but I appreciate the answers !

closed 2022-06-09 by 911GT2

davidplowman, 2022-06-09:
No problem. BTW, you can definitely ignore that particular "test pattern" error, I think we've suppressed it in libcamera now so it should disappear once we catch up with mainline again.