raspberrypi/picamera2 #174: [BUG]opencv_mertens_merge.py is not working.

TimBrownConsulting, 2022-06-16:
With the current version of Picamera2 (0.2.2) when I try to run the opencv_mertens_merge.py script the three images created appear to be the same.  It seems like setting the "ExposureTime" is not having any effect.

To reproduce,
Install the current release version of libcamera.
Install the latest version of picamera2 from git
Edit the script to output the three intermediate images - I added these three lines 

_cv2.imwrite("normal.jpg", normal)
cv2.imwrite("short.jpg", short)
cv2.imwrite("long.jpg", long)_

The images should be different but they are visually the same.

I am running on a 4GB Pi4 using Bullseye with an imx477 camera
It may not be relevant but the config.txt includes:
_dtoverlay=vc4-fkms-v3d_
Rather than the default:
_dtoverlay=vc4-kms-v3d_

With the default, when I disabled legacy camera support and performed the required reboot the HDMI port would be disabled.  So effectively enabling LibCamera disabled HDMI.  If I moved the SD card to a Pi3 then they would both work.  

I've tried running the script on a Pi3 with the same result.

Thanks in advance
Tim

davidplowman, 2022-06-17:
Hi, thanks for the report. I've tried this on a fresh Bullseye install with Picamera2 and OpenCV installed according to the instructions [here](https://github.com/raspberrypi/picamera2#installation). I added these two lines:
```
cv2.imwrite("short.jpg", short)
cv2.imwrite("long.jpg", long)
```
just below where it writes the "normal" image.

I'm seeing all the images coming out differently, as I would expect. I tried running Picamera2 from the code checked out of the repo but that was just the same. Also trying the "fkms" driver made no difference.

Are you able to post the exact example code that you are running? Would you be able to give this a try on a fresh install?

Also, if you run something like `libcamera-hello -t 0 --shutter 10000` does the window title bar report an exposure close to 10000 (probably 9994)?

Thanks!

TimBrownConsulting, 2022-06-18:
Hi David,

Thanks for the reply, I am absolutely sure this should work but It doesn't seem to be for my setup so:
I did as you suggested and created a new buster install using Raspberry Pi imager on my PC.
I tried booting it but it turned off the HDMI so I edited config.txt using my PC to change to vc4-fkms-v3d then I could see the screen.  
Completed the install as required by the installer
Installed picamera2 as per your instructions
Ran libcamera-hello and got exposure of 9999 - all working fine.
Installed Samba and gave the pi a static IP so I can copy files from my PC.
Made the user folders writeable
Copied the picamera2 folder from my PC to the pi (to get the examples folder).
Edited the exposure_filed.py to add these lines at the end:
__x = input("Enter")
picam2.stop()
picam2.start({"ExposureTime": 100000, "AnalogueGain": 2.0})
time.sleep(5)_
Couldn't notice any difference between the two images.
Edited the opencv_mertens_merge.py to add these two lines at the end:
_cv2.imwrite("short.jpg", short)
cv2.imwrite("long.jpg", long)_
Ran opencv_mertens_merge it and got the 4 attached images.
(It’s a microscope slide with a bad shading table)
I’ve attached a history log of the commands used, I’ve deleted my typo/error lines to save my embarrassment - Linux is not my first language.
Libcamera-hello still looks different (and reports different numbers) when run with different exposure numbers.
Looks like something I have done has broken the settings link from picamera2 to libcamera

I should note that this is quite an old Pi4 if that makes any difference.

I've also attached the edited version of the script.

[Attachments.zip](https://github.com/raspberrypi/picamera2/files/8932883/Attachments.zip)

TimBrownConsulting, 2022-06-19:
Update: I put this SD card in a pi3 based system and got the same results.

tvoverbeek, 2022-06-19:
@TimBrownConsulting You wrote: `to change to vc4-fkms-v3d`.
As far as I know picamera2 and libcamera require the real kms driver, not the fake one.
@davidplowman please correct me if this is not the case.

TimBrownConsulting, 2022-06-19:
@tvoverbeek I've just tried it on my pi3 setup with both the real and fake driver and both give the same (no difference between images) result.

davidplowman, 2022-06-20:
Thanks for attaching those files, I think I can see the problem.

In your version of the opencv_mertens_merge.py script you've passed the exposure/gain controls to the `start` function. In the published version it sets them using `set_controls`. I think there was a time when what you've done was correct, but there was a feeling in some discussions here a little while ago that it might be more consistent if controls were always set using `set_controls` and so I made the suggested changes.

So I'm sorry that you've been tripped up by this, but if you base your code off the currently published version then I think you will be OK.

To answer the other question, libcamera/Picamera2 work fine with both kms and fkms (fkms is obviously deprecated these days).

TimBrownConsulting, 2022-06-21:
Thank you for your help I knew I must have been doing something wrong.  Didn't occur to me to refresh the examples when I installed the latest code.  Thanks for the good work you and the team are doing.