raspberrypi/picamera2 #179: [BUG]

ketai-dhr, 2022-06-22:
**Describe the bug**
When I install dependencies on a card that has been used for a while, there will be a problem of not being able to import picamera2.

**To Reproduce**
https://github.com/raspberrypi/picamera2 According to the github README.md tutorial

**Expected behaviour**
I can't use picamera2, but whenever the system is reflashed, and executed again, it works fine. I'm 100% sure their environments are the same because I've been using the same thing.

**Console Output, Screenshots**
[image]

**Hardware :**
pi4(c03112) + 2022-04-07 bulleye armf + imx477

davidplowman, 2022-06-22:
Just to understand what's going on, you're saying you can flash a clean Bullseye, follow the instructions, and Picamera2 works OK. But then at some time later it stops working - is that correct?

From the call stack, it seems that it's complaining about the attempt to construct a Transform with no arguments, saying that there's only one constructor that takes an int. In fact, there should be _two_ constructors. One takes 4 ints (all optional), and one takes another Transform.

Is it possible that some other completely different Transform class has overwritten the original libcamera one?

ketai-dhr, 2022-06-22:
@davidplowman 
  I may have expressed incorrectly, on a clean Bullseye, picamera2 can be used after installing the dependencies. But it is very strange that our team has encountered that on a card that has been used for a while, it cannot be used after installing dependencies, and the problem is that picamera2 cannot be imported, and the same error message is reported.
   We have modified libcamera_dev and libcamera_apps, I suspect this may be the problem, but on a clean Bullseye, I also installed the modified libcamera and it can be used normally. That's the weird point.

davidplowman, 2022-06-22:
Yes, it's hard to understand what's happening. The Transform constructor definitions that you _should_ have are [here](https://github.com/raspberrypi/libcamera/blob/picamera2/src/py/libcamera/pymain.cpp#L550). There simply _isn't_ one that takes just an int. You should be able to test if things have gone wrong just by running
```
from libcamera import Transform
Transform()
```
If you installed your other dependencies one at a time and check at which point things go wrong, maybe that would give you a clue? Do any of these other dependencies define a Transform class?

ketai-dhr, 2022-06-22:
new:
[image]
[image]

old:
[image]
[image]

One of them is a new bulleye and the other is a bulleye that has been in use for a while. They are the same image, the same Raspberry Pi, the same environment

davidplowman, 2022-06-22:
In the two cases, is there a difference in the module search path? (`import sys` then look at `sys.path`), or in the `PYTHONPATH` environment variable?

ketai-dhr, 2022-06-23:
@davidplowman 
I can be sure that this is the same, our installation environment is unified with jenkins, and the steps must be exactly the same. And I'm also sure that we didn't use any python environment before using picamera2.

davidplowman, 2022-06-23:
I'm a bit at a loss really. At the moment I'm still suspecting that something else is coming along and behaving badly, but it's hard to know what to suggest as I can't reproduce this and debug it for you. Some more questions:

* Does `sys.modules` show anything? (Or perhaps `[v for k, v in sys.modules.items if 'libcamera' in k]`)
* What does the `Transform` class object look like? Maybe look at `dir(Transform)` - does that contain anything that might might help to identify what it is?

Sorry not to be more help.

davidplowman, 2022-08-17:
Hi, is there still anything we can help you with here? If there is please let us know, otherwise I'll consider closing this issue shortly.

closed 2022-08-19 by ketai-dhr