raspberrypi/picamera2 #180: [HOW-TO]  How to use the autofocus control

ketai-dhr, 2022-06-22:
**Describe what it is that you want to accomplish**
How to use the autofocus control.

**Describe alternatives you've considered**
In your C++ libcamera library, I remember that it supports autofocus control（AfTrigger）, but I can't find this control in the picamera2 packaged by it, can you tell me his name?

**Additional context**
[image]

tvoverbeek, 2022-06-22:
I guess you are using the Arducam autofocus camera.
They have their own modified libcamera and my guess is that AfTrigger is not exposed in their their bindings.
Their libcamera version might also be out of step with the RPi version.
Anyway AfTrigger is not available in the RPi libcamera version (at least it was not a few weeks ago when I last checked)

davidplowman, 2022-06-22:
Sorry, I'm afraid that's correct, autofocus is not currently available in the Raspberry Pi versions of these libraries.

ketai-dhr, 2022-06-23:
@davidplowman 
Well, thanks for your reply.

closed 2022-06-23 by ketai-dhr

surak, 2022-08-20:
When you have the libcamera from Arducam installed, this works in python: picam2.set_controls({"AfTrigger": 0})

millermuttu, 2023-04-16:
When we trigger Aftirigger using the below code,
'''
picam2.set_controls({"AfTrigger": 0})
'''

How do you wait until the autofocus happens or know that image is focused?

davidplowman, 2023-04-17:
This answer is specific to Raspberry Pi camera modules that support autofocus (namely the Camera Module 3). I don't know if it works with Arducam modules or software; if it doesn't you would need to go to Arducam for support.

After triggering an autofocus cycle you need to watch the `AfState` metadata so discover when it's finished. This is actually all done for you if you call `picam2.autofocus_cycle()`.