raspberrypi/picamera2 #182: [HOW-TO] h264 Timestamp File for mkv Conversion

ccostes, 2022-06-28:
I'm trying to convert the .h264 output files from this library to .mkv for playback. Is there any way to save a timestamp file like you can with the libcamera `--save-pts` option?

davidplowman, 2022-06-28:
At the moment this isn't possible. @chrisruk what do you think?

As an alternative/workaround you could do this:
```
from picamera2 import Picamera2
from picamera2.encoders import H264Encoder
from picamera2.outputs import FfmpegOutput

picam2 = Picamera2()
picam2.configure(picam2.video_configuration())
encoder = H264Encoder(bitrate=10000000)
output = FfmpegOutput('test.mkv')
picam2.start_recording(encoder, output)
```
though the timestamps in the file won't be as accurate.

chrisruk, 2022-06-29:
Good question, I'll have a look at this.

closed 2022-08-24 by chrisruk

will-v-pi, 2022-08-25:
@ccostes Hi, would it be useful for you if we added `# timestamp format v2` as the first line of the timestamp file (like the --save-pts option does for libcamera-vid) so you can import it straight into mkvmerge? If so I can add that feature