raspberrypi/picamera2 #189: [BUG] AttributeError: 'Picamera2' object has no attribute 'make_image_'

IanTBlack, 2022-06-29:
**Please only report one bug per issue!**

**Describe the bug**
capture_image fails at make_image_ on the current build (installed on 2022-06-29).

**To Reproduce**
capture_image('test.jpg')

**Expected behaviour**
You should receive the following error.
```
Exception in thread Thread-41:
Traceback (most recent call last):
  File "/usr/lib/python3.9/threading.py", line 954, in _bootstrap_inner
    self.run()
  File "/usr/lib/python3.9/threading.py", line 892, in run
    self._target(*self._args, **self._kwargs)
  File "/home/rdesc/.local/lib/python3.9/site-packages/picamera2/previews/null_preview.py", line 17, in thread_func
    callback(picam2)
  File "/home/rdesc/.local/lib/python3.9/site-packages/picamera2/previews/null_preview.py", line 40, in handle_request
    completed_request = picam2.process_requests()
  File "/home/rdesc/.local/lib/python3.9/site-packages/picamera2/picamera2.py", line 667, in process_requests
    if function():
  File "/home/rdesc/.local/lib/python3.9/site-packages/picamera2/picamera2.py", line 924, in <lambda>
    self.dispatch_functions([(lambda: self.make_image_(name))], signal_function)
AttributeError: 'Picamera2' object has no attribute 'make_image_'
```

**Hardware :**
Raspberry Pi 4B running RPi OS Bullseye 64 bit with PoE+ Hat and RPi HQ Camera.

**Additional context**
This issue is likely related to the make_image_ call, which has a preceding self.. It should probable be a child of the request module.

davidplowman, 2022-06-30:
Thanks for reporting this. I've pushed a fix here https://github.com/raspberrypi/picamera2/pull/185/commits/0fba7d84d129a41c75472ed265651c5e4c6a7820 which should get merged onto "next" shortly.

davidplowman, 2022-07-01:
The fix is merged onto "next" now, so I'll close this one. It will appear more generally when we make our next release. Thanks!

closed 2022-07-01 by davidplowman