raspberrypi/picamera2 #190: [BUG]

shineworld, 2022-06-30:
**Please only report one bug per issue!**

**Describe the bug**
FPS change, a lot, depending on object distance from the camera.
In the following video, you can see how the picam2.capture_array() slowdown when I change the object distance from the camera.
This does not happen with "libcamera-hello -t 0.

**To Reproduce**
The used code is:

```
#!/usr/bin/python3

import time

import cv2
import numpy as np
from picamera2 import Picamera2

picam2 = Picamera2()
main = {"format": 'RGB888', "size": (640, 480)}
raw = {"format": "SRGGB10_CSI2P", "size": (1332, 990)}
config = picam2.preview_configuration(main, raw=raw)
picam2.configure(config)
picam2.start()

frame_cnt = 0
frame_fps = 0
frame_info_show = False

tfps1 = time.perf_counter()
while True:
    try:
        try:
            frame = picam2.capture_array()
        except:
            frame = None
        if frame is None:
            print('image not ready!')
            continue
        cv2.imshow('frame', frame)
        cv2.waitKey(1)
        if frame_info_show:
            frame_info_show = False
            print('FPS = {:5.2f}'.format(frame_fps))
        frame_cnt += 1
        tfps2 = time.perf_counter()
        if (tfps2 - tfps1) > 1.0:
            tfps3 = time.perf_counter()
            frame_fps = frame_cnt / (tfps2-tfps1)
            frame_cnt = 1
            tfps1 = tfps3
            frame_info_show = True

    except Exception as e:
        print(e)

```

**Hardware :**
I'm using the latest Picamera2 and raspberry libcamera version on RPi 4B with IMX447 (motorized but not managed yet).

```
qem@raspberrypi:~ $ uname -a
Linux raspberrypi 5.15.32-v8+ #1538 SMP PREEMPT Thu Mar 31 19:40:39 BST 2022 aarch64 GNU/Linux

qem@raspberrypi:~ $ libcamera-hello --version
libcamera-apps build: 0dc5ea038c79 04-03-2022 (10:35:09)
libcamera build: v0.0.0+3544-22656360

qem@raspberrypi:~ $ pip list | grep picamera2
picamera2         0.2.2

```

shineworld, 2022-06-30:
WOW
I don't know that was possilbe to increase FPS till 60FPS.
With you suggestion FPS is now STABLE and very fast.
[image]

THANK you very much for help !

UcefMountacer, 2022-08-29:
> Perhaps the problem is just that the preview mode gives you a variable framerate? You can change it if you want:
> 
> ```
> picam2 = Picamera2()
> main = {"format": 'RGB888', "size": (640, 480)}
> raw = {"format": "SRGGB10_CSI2P", "size": (1332, 990)}
> controls = {"FrameDurationLimits": (1, 16667)}
> config = picam2.preview_configuration(main, raw=raw, controls=controls)
> picam2.configure(config)
> picam2.start()
> ```
> 
> The units are microseconds, so the above would let it vary between 60fps and the camera mode's maximum.

Hi,

I got thi error : 

```
config = cam.preview_configuration(main, raw=raw, controls=controls)
TypeError: 'CameraConfiguration' object is not callable
```

Tim-Brown-NZ, 2022-08-30:
Hi, the problem is that the code above was written for an older version of the library and the thread has not been updated.  
The preview_configuration method has been renamed create_preview_configuration.  
Rename your call and you should be fine.

UcefMountacer, 2022-08-30:
Thank you, that helped me !!