raspberrypi/picamera2 #191: [OTHER] Possible opencv example RGB/BGR mismatch

Boooke, 2022-06-30:
Irt the [opencv example here](https://github.com/raspberrypi/picamera2/blob/eeb992321242b2ea98195a002e5d9f819821c43f/examples/opencv_face_detect.py), should the line 
```
picam2.configure(picam2.preview_configuration(main={"format": 'XRGB8888', "size": (640, 480)}))
```
not be changed to (Note `format` value)
```
picam2.configure(picam2.preview_configuration(main={"format": 'XBGR8888', "size": (640, 480)}))
```
Or perhaps
```
picam2.configure(picam2.preview_configuration(main={"format": 'BGR8888', "size": (640, 480)}))
```
since as it is converted assuming BGR in a following line:
```
grey = cv2.cvtColor(im, cv2.COLOR_BGR2GRAY)
```

davidplowman, 2022-07-01:
Hi, indeed these are always good questions!

I think we inherit these format names from _libcamera_, but there are some interesting conventions at play in the Linux world. They seem to take the endianness of the platform into account which really confuses me. Anyhow, it means that what is called `XRGB8888` actually ends up with every pixel looking like `[B, G, R, X]` in a _numpy_ array, which I believe is what _OpenCV_ expects. I suppose the critical question is: does it work? I agree that one could have a debate as to whether to rename all the formats (which would require some extra look-up tables in our code)...

As regards the 24-bit format, you're right, that would probably be better in this example, though it works as it is. _Picamera2_ quite often defaults to 32-bit formats because those work with GLES accelerated preview window (and the 24-bit ones don't), but if you aren't using that then the 24-bpp formats are fine too. Some _OpenCV_ operations require 3-channel images, but others will accept 4-channel ones quite happily.

Boooke, 2022-07-01:
Thanks, David, for the detailed and quick answer! Alright - I will follow the practice of this example, and try the other formats out if there are any problems.

closed 2022-07-01 by Boooke