raspberrypi/picamera2 #192: [BUG] Error when following install instructions for python3-libcamera python3-kms++

Bluesguy1984, 2022-07-01:
**Please only report one bug per issue!**

**Describe the bug**
When I follow the instructions for the installation I get error messages such as:

- dpkg: error processing archive
- dpkg-deb: error: paste subprocess was killed by signal (Broken pipe)
See below for more details in the console output.

**To Reproduce**
sudo apt install -y python3-libcamera python3-kms++

**Expected behaviour**
The software installs without any errors

**Console Output, Screenshots**

```
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
python3-kms++ is already the newest version (0~git20211110~105525+2236a8c-1).
The following packages were automatically installed and are no longer required:
  libfuse2 libva-wayland2
Use 'sudo apt autoremove' to remove them.
The following NEW packages will be installed:
  python3-libcamera
0 upgraded, 1 newly installed, 0 to remove and 0 not upgraded.
Need to get 0 B/160 kB of archives.
After this operation, 780 kB of additional disk space will be used.
(Reading database ... 165170 files and directories currently installed.)
Preparing to unpack .../python3-libcamera_0~git20220510+22656360-1_armhf.deb ...
Unpacking python3-libcamera (0~git20220510+22656360-1) ...
dpkg: error processing archive /var/cache/apt/archives/python3-libcamera_0~git20
220510+22656360-1_armhf.deb (--unpack):
 trying to overwrite '/usr/lib/python3/dist-packages/libcamera/__init__.py', whi
ch is also in package libcamera-dev 0.0.7
dpkg-deb: error: paste subprocess was killed by signal (Broken pipe)
Errors were encountered while processing:
 /var/cache/apt/archives/python3-libcamera_0~git20220510+22656360-1_armhf.deb
E: Sub-process /usr/bin/dpkg returned an error code (1)
```

**Hardware :**
Raspberry PI 4 Model B Rev 1.2 using the Arducam 64mp Hawkeye Camera

**Additional context**
If I use the following command:
`libcamera-vid -t 0 --autofocus`

it works perfectly.

davidplowman, 2022-07-01:
Hi, I wonder if it might be because you've installed a proprietary Arducam version of libcamera. I believe they supply their own version to add the autofocus functionality but I guess this wouldn't be compatible with Picamera2. You might consider asking them for advice.

davidplowman, 2022-07-06:
I'm going to close this as I'm as certain as I can be that this is caused by using the Arducam proprietary software. If you're able to reproduce this using the officially supported software, or other information comes to light to demonstrate that there is a general problem, then please open a new issue. Thanks!

closed 2022-07-06 by davidplowman