raspberrypi/picamera2 #196: [DOCUMENTATION]

andrum99, 2022-07-05:
**This form is ONLY for reporting issues with https://datasheets.raspberrypi.com/camera/picamera2-manual-draft.pdf**

**Describe the problem**
Section 2.1 on page 5 says `Some lower-power devices, such as Pi Zeroes,...`. This should probably be reworded to state the actual product name, for example:

`Some lower-power devices, such as Raspberry Pi Zero,...`

andrum99, 2022-07-05:
Just to say - I've ben adding a separate issue for each thing I come across, per the instructions in the issue template, but if you would prefer such copy edit-related issues to be all in one issue then please let me know. Also, if you would prefer I don't submit copy edit-type issues, and leave it to internal copy editors, then that's fine too - just let me know.

davidplowman, 2022-07-06:
Thanks, I've fixed this and it will be available in the next release.

(I'm fine to receive corrections, especially glaring typos! Obviously this documentation hasn't been through the formal process at this end yet, but that will happen in due course. In the meantime, I'm keen to "have something out there" to help folks get started.)

aallan, 2022-07-06:
@andrum99 While glaring typos are okay, its probably best not to try and copy edit the document. The point of the being DRAFT watermark is to indicate that (amongst other things) the document has not gone through the copy edit process. These sorts of smaller CE issues will get picked up when that happens.

andrum99, 2022-07-06:
> @andrum99 While glaring typos are okay, its probably best not to try and copy edit the document. The point of the being DRAFT watermark is to indicate that (amongst other things) the document has [...]

Thanks - I realised that would probably be the case, eventually. I will hold my fire in future.