raspberrypi/picamera2 #207: [DOCUMENTATION] Cannot install picamera2 using DRAFT documentation

baqwas, 2022-07-05:
**This form is ONLY for reporting issues with https://datasheets.raspberrypi.com/camera/picamera2-manual-draft.pdf**

**Please only report one documentation bug per issue!**

**Describe the problem**
Using the instructions in [The Picamera2 Library](https://datasheets.raspberrypi.com/camera/picamera2-manual-draft.pdf#%5B%7B%22num%22%3A3%2C%22gen%22%3A0%7D%2C%7B%22name%22%3A%22XYZ%22%7D%2C0%2C841.89%2Cnull%5D) on Page 5 Section 2.2, I am stumbling at the final step for the installation of **picamera2**: 
`sudo apt install -y python3-picamera2`

Used cut-n-paste from the Picamera2 documentation; checked for obvious typos but cannot see anything obvious. The output message(s) from the command are:
`Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
E: Unable to locate package python3-picamera2`

**Suggested Improvement**
Can you suggest a fix for the problem or better wording?
Installed the prerequisites (two installs + numpy upgrade) without any significant errors. I may have overlooked some other steps (_perhaps_ documented elsewhere) but I suspect that this is an operator _self-inflicted_ error because I'm failing at the starting block.

**Hardware/Software :**
Is the problem with any particular type of Pi or version of the software?
My platform is:
`$ uname -a
Linux localhost 5.15.32-v7+ #1538 SMP Thu Mar 31 19:38:48 BST 2022 armv7l GNU/Linux
`
The OS image is from the Foundation's site:
`Raspbian GNU/Linux 11 (bullseye) armv`
on
`Raspberry Pi Zero 2 W Rev 1.0`

As customary, **apt update/upgrade** was performed before embarking on the installation of **picamera2**.

**Additional context**
Don't have any alternate RPi to perform parallel tests. (Waiting for RPi4 8GB since late August 2021 across the pond). Of course, have unpacked HQ camera and both lenses but want to test with V2 camera before attempting anything more ambitious. Thanks.

Kind regards.

davidplowman, 2022-07-06:
Sorry, the instructions in the manual are a little bit ahead of reality at the moment, we haven't got an apt package yet. For now, please continue to use the instructions [in this repository](https://github.com/raspberrypi/picamera2#installation). Obviously that will get fixed soon. Thanks.

closed 2022-07-06 by davidplowman

aguaviva, 2022-09-14:
I am still hitting this one. Are there any updates?

davidplowman, 2022-09-15:
Hi, just to confirm: it's still not finding the python3-picamera2 package on apt? The package is definitely posted now, could you verify that you've done `sudo apt update`? Thanks.

Note also that if you've previously installed Picamera2 using pip, then we advise uninstalling it first (`pip3 uninstill picamera2`).

baqwas, 2022-09-15:
Thanks, @davidplowman.

Here is the output from just a few minutes ago:

```
pi@raspbari17:~/projects $ sudo apt update && sudo apt -y upgrade
Get:1 http://raspbian.raspberrypi.org/raspbian bullseye InRelease [15.0 kB]
Get:2 http://archive.raspberrypi.org/debian bullseye InRelease [23.7 kB]
Get:3 http://archive.raspberrypi.org/debian bullseye/main armhf Packages [301 kB]
Fetched 340 kB in 2s (141 kB/s)   
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
All packages are up to date.
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
Calculating upgrade... Done
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
pi@raspbari17:~/projects $ sudo apt install picamera2
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
E: Unable to locate package picamera2
pi@raspbari17:~/projects $ uname -a
Linux raspbari17 5.15.61-v7+ #1579 SMP Fri Aug 26 11:10:59 BST 2022 armv7l GNU/Linux
pi@raspbari17:~/projects $ 

```

I am still a newbie with respect picamera2 - running very, very simple tutorials. Thanks again for all your work in making picamera2 more accessible.

Kind regards.

aguaviva, 2022-09-15:
It seems so, here is what I am getting:

```
user@pulga:~ $ pip3 uninstall picamera2
WARNING: Skipping picamera2 as it is not installed.

user@pulga:~ $ sudo apt update
Get:1 http://archive.raspberrypi.org/debian buster InRelease [32.6 kB]
Get:2 http://raspbian.raspberrypi.org/raspbian buster InRelease [15.0 kB]
Get:3 http://raspbian.raspberrypi.org/raspbian buster/main armhf Packages [13.0 MB]
Get:4 http://archive.raspberrypi.org/debian buster/main armhf Packages [392 kB]
Fetched 13.5 MB in 27s (504 kB/s)
Reading package lists... Done
Building dependency tree
Reading state information... Done
2 packages can be upgraded. Run 'apt list --upgradable' to see them.

user@pulga:~ $ sudo apt install -y python3-picamera2
Reading package lists... Done
Building dependency tree
Reading state information... Done
E: Unable to locate package python3-picamera2
```

tvoverbeek, 2022-09-15:
@aguaviva You are on buster. python3-picamera2 is only available on bullseye

@baqwas Package name is python3-picamera2, not picamera2

chrisruk, 2022-09-15:
Yeah I was just checking that, the deb package does only appear in the bullseye repo -

```
curl https://archive.raspberrypi.org/debian/dists/bullseye/main/binary-armhf/Packages | grep -i "python3-picamera2"
```

No package is found for the buster repo -

```
curl https://archive.raspberrypi.org/debian/dists/buster/main/binary-armhf/Packages | grep -i "python3-picamera2"
```

davidplowman, 2022-09-15:
Sorry, that wasn't very clear. It means should work on other hardware platforms (Pi 4Bs, Pi 3s etc., compute modules, and so on). It does not apply to older software releases, and I will get that fixed. It is perhaps stated more clearly in the "Getting Started" chapter of the manual.

baqwas, 2022-09-15:
This note is simply to add depth to my previous comment:

```
  `.::///+:/-.        --///+//-:``    pi@raspbari17 
 `+oooooooooooo:   `+oooooooooooo:    OS: Raspbian GNU/Linux 11 (bullseye) armv 
  /oooo++//ooooo:  ooooo+//+ooooo.    Host: Raspberry Pi Zero 2 W Rev 1.0 
  `+ooooooo:-:oo-  +o+::/ooooooo:     Kernel: 5.15.61-v7+ 
   `:oooooooo+``    `.oooooooo+-      Uptime: 6 days, 20 hours, 59 mins 
     `:++ooo/.        :+ooo+/.`       Packages: 1939 (dpkg) 
        ...`  `.----.` ``..           Shell: bash 5.1.4 
     .::::-``:::::::::.`-:::-`        CPU: BCM2835 (4) @ 1.000GHz 
    -:::-`   .:::::::-`  `-:::-       Memory: 110MiB / 363MiB 
   `::.  `.--.`  `` `.---.``.::`      CPU Usage: 14% 
       .::::::::`  -::::::::` `       Disk (/): 23G / 59G (40%) 
 .::` .:::::::::- `::::::::::``::.    Local IP: 172.16.1.133 
-:::` ::::::::::.  ::::::::::.`:::-   Users: pi 
::::  -::::::::.   `-::::::::  ::::
-::-   .-:::-.``....``.-::-.   -::-                           
 .. ``       .::::::::.     `..`..                            
   -:::-`   -::::::::::`  .:::::`
   :::::::` -::::::::::` :::::::.
   .:::::::  -::::::::. ::::::::
    `-:::::`   ..--.`   ::::::.
      `...`  `...--..`  `...`
            .::::::::::
             `.-::::-`

```