raspberrypi/picamera2 #212: [BUG] applied AWB preset stays after reconfiguration

arminveres, 2022-07-07:
**Describe the bug**
Disabling the auto white balance after having it initially enabled, with a preset, does not reset the preset, and possibly also leaves the AWB on.

**To Reproduce**
```python
#!/usr/bin/python3

import time
import libcamera
from picamera2 import Picamera2, Preview

picam2 = Picamera2()
picam2.start_preview(Preview.QT)

preview_config = picam2.create_preview_configuration(
    controls={
        "AwbEnabled": 1, 
        "AwbMode": libcamera.controls.AwbModeEnum.Cloudy
    }
)
picam2.configure(preview_config)

picam2.start()
time.sleep(1)

picam2.set_controls({"AwbEnable": 0})
time.sleep(5)
```

**Expected behaviour**
I expect for the returned images to be without any presets and AWB.
With the Cloudy preset, it is most obvious, at least in my environment, that the preset stays.

**Hardware :**
- Raspberry Pi 4, 4G
- Camera Module v2.1

**Additional context**
This effect also persists, if the camera is stopped and reconfigured.

arminveres, 2022-07-08:
Hello, thanks for answering.

Let me reformulate what I am intending. 
I would like to be able to remove any preset that has been set before, e.g. enabling AWB and choosing the `Cloudy` preset, and then disable the AWB and remove any effect/preset had, so basically a clean slate.

I also tried starting and stopping the camera, to no avail. (I am guessing the AWB values get reset after disacquiring or before acquiring the camera).

One use case I am trying to achieve is to apply a manual frame by frame colour temperature, which works by itself, only the image stays 'tainted' by one of the presets.

waisaacs, 2022-07-08:
Is the typo "AwbEnabled": 1 relevant here?  The control is "AwbEnable" unless I'm not finding some alias hidden somewhere.  But it sounds like AwbMode gets set and sticks independent of AwbEnable?

davidplowman, 2022-07-09:
Hi, I guess I'm still not quite sure I understand, but let me try and explain how things work:

AWB controls the red and blue colour gains. You can find them in the image metadata (`picam2.capture_metadata()["ColourGains"]`). While the AWB is running ("enabled") you can change its mode, for example to cloudy (`picam2.set_controls({"AwbMode": libcamera.controls.AwbModeEnum.Cloudy})`) and you can also set it back to normal (`picam2.set_controls({"AwbMode": libcamera.controls.AwbModeEnum.Auto})`).

You can also disable the AWB completely (`picam2.set_controls({"AwbEnable": 0})`) at which point the colour gains will never change by themselves. You can set the colour gains (`picam2.set_controls({"ColourGains": (1, 1)})`), or in fact, setting the colour gains will disable the AWB automatically anyway.

When you re-enable the AWB (`picam2.set_controls({"AwbEnable": 1})`) it will return to whatever the AWB mode was previously (either `Cloudy`, `Auto` or whatever). One thing that might seem confusing is that setting an AWB mode does _not_ disable AWB, it merely restricts the range of possibilities that it searches, but it's still running.

All this state is persistent across everything except stopping and restarting the entire camera system. So camera start/stop/configure will have no effect on any AWB settings (unless you explicitly put AWB controls into your camera modes, but that would be your choice).

I hope that's a bit clearer. If not, perhaps try to include an exact script and say at the end exactly how the AWB setting of the captured image is not as you expect, then we can look at that. Thanks!

arminveres, 2022-07-11:
Hi again,

Thanks for the explanation and sorry for any confusion. Did not know of the gain values and their relevance for the AWB, that was exactly what I have been looking for.

Getting the metadata when AWB is disabled, returns a tuple, which I am guessing are the default values of the camera.
The tuple value, in my case `(1.4193880558013916, 1.7060798406600952)`, I am getting seems to be persistent across different calls.
So in a sense, setting/applying those numbers, after having AWB enabled with an AWB preset, should get me a 'raw' image, i.e. one with the default values.

As you basically said, my issue was, that I was not aware, where those 'gains' were getting applied. So my intention is/was to set the colours back to a state, where no AWB/AWB preset is getting applied.

Unless I am wrong, please correct me if I am, I think this issue is resolved.
Thanks!

I am achieving that currently with the following script:
```python
#!/usr/bin/python3

import time
import libcamera

from picamera2 import Picamera2, Preview

picam2 = Picamera2()
picam2.start_preview(Preview.QT)

preview_config = picam2.create_preview_configuration(
    controls={
        "AwbEnable": 1,
        "AwbMode": libcamera.controls.AwbModeEnum.Cloudy,
    }
)
picam2.configure(preview_config)

picam2.start()
# print(picam2.capture_metadata())
time.sleep(5)
picam2.set_controls({"ColourGains": (1.4193880558013916, 1.7060798406600952)})
# print(picam2.capture_metadata())
time.sleep(5)
# print(picam2.capture_metadata())

```

davidplowman, 2022-07-26:
Thanks, closing this one. Please open a new issue if you have other questions/problems.

closed 2022-07-26 by davidplowman