raspberrypi/picamera2 #217: [BUG] Initialization failure after Pi update

ccostes, 2022-07-10:
I recently let Raspbian install updates (believe this runs `packagekit role='update-packages'`) and picamera2 now fails to initialize. Any help would be much appreciated!

Repro is quite simple, from IDLE:
```
Python 3.9.2 (default, Mar 12 2021, 04:06:34)
[GCC 10.2.1 20210110] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> from picamera2 import Picamera2
>>> picam2 = Picamera2()
[0:17:47.160732228] [1643]  INFO Camera camera_manager.cpp:293 libcamera v0.0.0+3700-f30ad033
[0:17:47.235985561] [1644]  INFO RPI raspberrypi.cpp:1368 Registered camera /base/soc/i2c0mux/i2c@1/ov5647@36 to Unicam device /dev/media3 and ISP device /dev/media0
2022-07-10T02:03:31.595Z | ERROR    | Camera __init__ sequence did not complete.
Traceback (most recent call last):
  File "/home/pi/.local/lib/python3.9/site-packages/picamera2/picamera2.py", line 77, in __init__
    self.open_camera()
  File "/home/pi/.local/lib/python3.9/site-packages/picamera2/picamera2.py", line 178, in open_camera
    if self.initialize_camera():
  File "/home/pi/.local/lib/python3.9/site-packages/picamera2/picamera2.py", line 162, in initialize_camera
    self.sensor_resolution = self.camera.properties["PixelArraySize"]
KeyError: 'PixelArraySize'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/home/pi/.local/lib/python3.9/site-packages/picamera2/picamera2.py", line 81, in __init__
    raise RuntimeError("Camera __init__ sequence did not complete.")
RuntimeError: Camera __init__ sequence did not complete.
```

ccostes, 2022-07-11:
Added some debug logging to `picamera2.py` to print out `self.camera.properties` - it does appear to have a `PixelArraySize` field, but the keys are `libcamera.ControlId`s and not strings.
```
{
    libcamera.ControlId(3, Model, ControlType.String): 'ov5647', 
    libcamera.ControlId(4, UnitCellSize, ControlType.Size): libcamera.Size(1400, 1400), 
    libcamera.ControlId(1, Location, ControlType.Integer32): 2, 
    libcamera.ControlId(2, Rotation, ControlType.Integer32): 0, 
    libcamera.ControlId(7, PixelArrayActiveAreas, ControlType.Rectangle): libcamera.Rectangle(16, 6, 2592, 1944), 
    libcamera.ControlId(5, PixelArraySize, ControlType.Size): libcamera.Size(2592, 1944), 
    libcamera.ControlId(10, ColorFilterArrangement, ControlType.Integer32): 2, 
    libcamera.ControlId(8, ScalerCropMaximum, ControlType.Rectangle): libcamera.Rectangle(0, 0, 0, 0)
}
```

chrisruk, 2022-07-11:
Do you mind running:

```
pip3 show picamera2
```

The version should be 0.2.3

If it's not, could you do:

```
pip3 install picamera2 -U
```

ccostes, 2022-07-11:
Ah yep, I was still on 0.2.2 and the upgrade command fixed it. I had tried `pip3 install picamera2` but I guess that doesn't check for updates.

Thanks!