raspberrypi/picamera2 #218: [BUG] Raspberry Pi 4: Cannot import name 'Size' or 'ControlType' from libcamera depending on version of libcamera

swordsmanluke, 2022-07-10:
**Describe the bug**
I can't seem to import from picamera2 regardless of the libcamera version I'm using. 

When I enable "Legacy Camera Support" in raspi-config, picamera2 fails to import 'Size' from libcamera. 
When I install a more recent libcamera, picamera2 fails to import 'ControlType'.

I also checked the libcamera API documentation (https://www.libcamera.org/api-html/classes.html) and I don't see a ControlType class?

Is there a recommended version of libcamera to install? 

**To Reproduce**
Raspberry Pi 4 (4 GB RAM model)
Raspberry PI OS Lite (32 bit, Buster-base Release 2022-04-04)

Sample program:
```
from picamera2 import PiCamera
# That's all that's required
```

Install picamera2 with no gui according to workaround here: https://github.com/raspberrypi/picamera2/issues/184
In `sudo raspi-config` Interface Options -> Legacy Camera, enable Legacy Camera suport. Reboot

Run sample program. Note "cannot import name 'Size'" error

Install libcamera from wheel: https://github.com/ArduCAM/RaspberryPi/blob/master/Motorized_Focus_Camera/python/libcamera-1.0.2-cp39-cp39-linux_armv7l.whl
`python3 -m pip install ./libcamera-1.0.2-cp39-cp39-linux_armv7l.whl `

Run sample program. Note "cannot import name 'ControlType'"

**Expected behaviour**
importing PiCamera works

**Console Output, Screenshots**

With "legacy camera"

```
Traceback (most recent call last):
  File "/home/pi/RaspberryPi/Motorized_Focus_Camera/python/picam2.py", line 1, in <module>
    from picamera2 import PiCamera
  File "/home/pi/.local/lib/python3.9/site-packages/picamera2/__init__.py", line 1, in <module>
    from .configuration import CameraConfiguration, StreamConfiguration
  File "/home/pi/.local/lib/python3.9/site-packages/picamera2/configuration.py", line 1, in <module>
    from .controls import Controls
  File "/home/pi/.local/lib/python3.9/site-packages/picamera2/controls.py", line 4, in <module>
    from libcamera import ControlType, Size, Rectangle
ImportError: cannot import name 'Size' from 'libcamera' (/usr/lib/python3/dist-packages/libcamera/__init__.py)
```

With more recent libcamera:

```
Traceback (most recent call last):
  File "/home/pi/RaspberryPi/Motorized_Focus_Camera/python/./Autofocus.py", line 5, in <module>
    from RpiCamera import Camera
  File "/home/pi/RaspberryPi/Motorized_Focus_Camera/python/RpiCamera.py", line 10, in <module> 
    from picamera2 import Picamera2, Preview
  File "/home/pi/.virtualenvs/moto_cam/lib/python3.9/site-packages/picamera2/__init__.py", line 1, in <module>
    from .configuration import CameraConfiguration, StreamConfiguration
  File "/home/pi/.virtualenvs/moto_cam/lib/python3.9/site-packages/picamera2/configuration.py", line 1, in <module>
    from .controls import Controls
  File "/home/pi/.virtualenvs/moto_cam/lib/python3.9/site-packages/picamera2/controls.py", line 4, in <module>
    from libcamera import ControlType, Size, Rectangle
ImportError: cannot import name 'ControlType' from 'libcamera' (/home/pi/.virtualenvs/moto_cam/lib/python3.9/site-packages/libcamera.cpython-39-arm-linux-gnu

**Hardware :**
Raspberry Pi 4
Arducam 5MP (Repros with RaspiCam v1.3 as well)

**Additional context**
With Legacy Camera support enabled, 
```
$ sudo ldconfig -v | grep camera
        libcamera-base.so.0 -> libcamera-base.so.0.0.0
--
        libcamera.so.0 -> libcamera.so.0.0.0
```

With it disabled, but installing a wheel:
```
$ sudo ldconfig -v | grep camera
$
```

ccostes, 2022-07-11:
Looks like I'm seeing the same issue in https://github.com/raspberrypi/picamera2/issues/217

shineworld, 2022-07-11:
What camera version do you use ?

I've either Arducam Motorized Cameras imx219 (SKU: B0393) and imx477 (SKU: B0272)
and all works perfectly, overall with the latest versions:

```
qem@raspberrypi:~ $ uname -a
Linux raspberrypi 5.15.32-v8+ #1538 SMP PREEMPT Thu Mar 31 19:40:39 BST 2022 aarch64 GNU/Linux
qem@raspberrypi:~ $ libcamera-hello --version
libcamera-apps build: 35266e84043f 07-07-2022 (17:59:45)
libcamera build: v0.0.0+3700-f30ad033
qem@raspberrypi:~ $ pip list | grep picamera2
picamera2                        0.2.3
```

swordsmanluke, 2022-07-11:
Thanks! Here's my picamera2 version: 0.2.3

```
$ pip3 show picamera2
Name: picamera2
Version: 0.2.3
Summary: The libcamera-based Python interface to Raspberry Pi cameras, based on the original Picamera library
Home-page: https://github.com/RaspberryPi/picamera2
Author: Raspberry Pi & Raspberry Pi Foundation
Author-email: picamera2@raspberrypi.com
License: BSD 2-Clause License
Location: /home/pi/.local/lib/python3.9/site-packages
Requires: pillow, pyopengl, python-prctl, numpy, PyQt5, v4l2-python3, simplejpeg, piexif, PiDNG
Required-by: 
```

Trying to apt-install libcamera yields:
```
$ sudo apt install -y python3-libcamera        
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
python3-libcamera is already the newest version (0~git20220705+f30ad033-1).
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
```

With both of those confirmed, I still get 
```
$ python -c 'import picamera2'
Traceback (most recent call last):
  File "<string>", line 1, in <module>
  File "/home/pi/.local/lib/python3.9/site-packages/picamera2/__init__.py", line 1, in <module>
    from .configuration import CameraConfiguration, StreamConfiguration
  File "/home/pi/.local/lib/python3.9/site-packages/picamera2/configuration.py", line 1, in <module>
    from .controls import Controls
  File "/home/pi/.local/lib/python3.9/site-packages/picamera2/controls.py", line 4, in <module>
    from libcamera import ControlType, Size, Rectangle
ImportError: cannot import name 'ControlType' from 'libcamera' (/home/pi/.local/lib/python3.9/site-packages/libcamera.cpython-39-arm-linux-gnueabihf.so)
```

shineworld, 2022-07-12:
This is my list of commands JUST after installing a clean Raspberry Pi4 Bulleyes 64-bit:

```
> sudo apt-get update
> sudo apt-get upgrade
> sudo apt-get full-upgrade
> sudo reboot

> sudo nano /boot/config.txt

   for IMX219:
    # Automatically load overlays for detected cameras
    camera_auto_detect=0
    dtoverlay=imx219

  for IMX477:
    # Automatically load overlays for detected cameras
    camera_auto_detect=0
    dtoverlay=imx477

> sudo reboot
``` 
 
 At this point libcamera-apps are already installed and working:

```
> libcamera-hello --version
libcamera-apps build: 35266e84043f 07-07-2022 (17:59:45)
libcamera build: v0.0.0+3700-f30ad033
```

Follow activity to enable camera focus motor management (if camera support it because motorized):

```
> sudo modprobe i2c-dev
> sudo nano /boot/config.txt

Add at end:

    [all]
    dtparam=i2c_vc=on

> sudo reboot
```

Packages to install to have Picamera2, OpenCV working:

```
> sudo apt-get update
> sudo apt-get upgrade
> sudo apt-get autoremove
    
> sudo apt install -y python3-libcamera python3-kms++
> sudo apt install -y python3-pyqt5 python3-prctl libatlas-base-dev ffmpeg python3-pip
> pip3 install numpy --upgrade
> pip3 install picamera2
    
> sudo apt install -y python3-opencv
> sudo apt install -y opencv-data
```

chrisruk, 2022-07-12:
@swordsmanluke I think your python interpreter is finding the arducam version of libcamera by the looks of it still.  Can you uninstall that.

tvoverbeek, 2022-07-12:
As far as I know the Arducam version of libcamera installs in /usr/local/lib and the 'official' apt version in /usr/lib.
Remove the one in /usr/local/lib since /usr/local/lib has preference over /usr/lib

gunderbolt, 2022-07-13:
@shineworld: You are correct about the Legacy camera support. It is hinted at in the [Raspberry Pi Camera Documentation](https://www.raspberrypi.com/documentation/accessories/camera.html#libcamera-and-the-legacy-raspicam-camera-stack). and mentioned in [this linked RPi news post](https://www.raspberrypi.com/news/bullseye-camera-system/). Particularly, it implies that one needs to stick with Picamera if using the legacy stack:

> An alternative [Python interface] to the old Picamera, imaginatively named Picamera2, is in development. This too will integrate much more directly with established Python libraries to access things like windowing and graphics functions. Picamera2 will be developed by Raspberry Pi (unlike Picamera itself, which is actually third-party code), and this will facilitate both support and continuing future development.

Is there a way Picamera2 can detect if the legacy option is enabled or use this Size error to give the user a meaningful error message?

swordsmanluke, 2022-07-14:
@shineworld 
THANK YOU SO MUCH. I blew away my install and started over from a 64-bit OS.
Your instructions worked for me!

I'm actually using [this camera](https://manuals.plus/arducam/b0176-5mp-camera-module-for-raspberry-pi-manual) so I put  "ov5647" in boot/config.txt
```bash
    camera_auto_detect=0
    dtoverlay=ov5647
```

I never would have gotten any of this figured out without your detailed instructions. Thank you.

closed 2022-07-14 by swordsmanluke

trezero, 2022-09-02:
@shineworld I have been fighting to get my camera working for the past 3 weeks and your instructions are the first and only ones that actually worked. Thank you SOO Much! I wish I could buy you a beer (or 5!). Truly appreciate you.

trezero, 2022-09-04:
One last question on this though... Now that I finally have libcam working with my pi. It seems that I have lost the ability to do Autofocus. The Autofocus capability of the Arducam seems to come from libcamera_dev. But if I install libcamera_dev after everything else above, it breaks everything. Any ideas how to get autofocus working? I'm working with the IMX519 FYI.

Sarac7t, 2022-11-28:
Hai guys I am start up in Raspberry Pi

I have bought ArduCAM PTZ-IMX477 12MP (B016712MP) camera and Raspberry pi 4B 2GB last week.
The camera looks assume and the installation also easy, but on the software side i have facing the issue like

"ImportError: cannot import name 'ControlType' from 'libcamera' (/home/system/.local/lib/python3.9/site-packages/libcamera.cpython-39-arm-linux-gnueabihf.so)"

I have followed the Arducam official site procedure but I have facing this error..
https://www.arducam.com/docs/cameras-for-raspberry-pi/ptz-camera/12mp-imx477-ptz-high-quality-camera/

And I have tried libcamera-still and libcamera-hello i can able to open the camera but unable to focus manually...

Just receiving the same blur image on all commands.

On the hardware segment i have check manytimes and also test the camera's servo motor by twisting slowly in my hand and libcamera-still the servo is working..

system@raspberrypi:~/PTZ-Camera-Controller $ libcamera-still
Made X/EGL preview window
[1:30:09.307289551] [5005]  INFO Camera camera_manager.cpp:293 libcamera v0.0.1+54-d528119f
[1:30:09.363814600] [5007]  INFO RPI raspberrypi.cpp:1414 Registered camera /base/soc/i2c0mux/i2c@1/imx477@1a to Unicam device /dev/media3 and ISP device /dev/media2
[1:30:09.366294897] [5005]  INFO Camera camera.cpp:1026 configuring streams: (0) 2028x1520-YUV420
[1:30:09.367260345] [5007]  INFO RPI raspberrypi.cpp:800 Sensor: /base/soc/i2c0mux/i2c@1/imx477@1a - Selected sensor format: 2028x1520-SBGGR12_1X12 - Selected unicam format: 2028x1520-pBCC
system@raspberrypi:~/PTZ-Camera-Controller $ vcgencmd get_camera
supported=0 detected=0, libcamera interfaces=0

system@raspberrypi:~/PTZ-Camera-Controller $ libcamera-hello --list-cameras
Available cameras
-----------------
0 : imx477 [4056x3040] (/base/soc/i2c0mux/i2c@1/imx477@1a)
    Modes: 'SRGGB10_CSI2P' : 1332x990 [120.05 fps - (696, 528)/2664x1980 crop]
           'SRGGB12_CSI2P' : 2028x1080 [50.03 fps - (0, 440)/4056x2160 crop]
                             2028x1520 [40.01 fps - (0, 0)/4056x3040 crop]
                             4056x3040 [10.00 fps - (0, 0)/4056x3040 crop]

Can pls help me out from the issue,....