raspberrypi/picamera2 #221: [BUG]

jenyak, 2022-07-16:
After install Raspbian updates. picamera2 now fails to initialize.
I also run 
```
sudo apt install -y python3-libcamera python3-kms++
sudo apt install -y python3-pyqt5 python3-prctl libatlas-base-dev ffmpeg python3-pip
pip3 install numpy --upgrade
pip3 install picamera2
```

**To Reproduce**
```
import time

from picamera2 import Picamera2, Preview

picam2 = Picamera2()
picam2.start_preview(Preview.QTGL)

preview_config = picam2.create_preview_configuration()
picam2.configure(preview_config)

picam2.start()
time.sleep(5)
```

**Console Output, Screenshots**
```
$ python test.py 
Traceback (most recent call last):
  File "/home/pi/test.py", line 3, in <module>
    from picamera2 import Picamera2, Preview
  File "/home/pi/.local/lib/python3.9/site-packages/picamera2/__init__.py", line 1, in <module>
    from .configuration import CameraConfiguration, StreamConfiguration
  File "/home/pi/.local/lib/python3.9/site-packages/picamera2/configuration.py", line 1, in <module>
    from .controls import Controls
  File "/home/pi/.local/lib/python3.9/site-packages/picamera2/controls.py", line 4, in <module>
    from libcamera import ControlType, Size, Rectangle
  File "/usr/lib/python3/dist-packages/libcamera/__init__.py", line 4, in <module>
    from ._libcamera import *
ImportError: /usr/lib/python3/dist-packages/libcamera/_libcamera.cpython-39-arm-linux-gnueabihf.so: undefined symbol: _ZN9libcamera8controls7AfRangeE
```

**Hardware :**
Pi 4b with HQ camera

**Additional context**
```
$ uname -a
Linux 5.15.32-v7l+ #1538 SMP Thu Mar 31 19:39:41 BST 2022 armv7l GNU/Linux
```
```
$ python -V
Python 3.9.2
```
```
apt show python3-libcamera 
Package: python3-libcamera
Version: 0~git20220705+f30ad033-1
```
```
 $ libcamera-hello --version
libcamera-apps build: 35266e84043f 07-07-2022 (17:54:29)
libcamera build: v0.0.0+3495-bd38def2
```

```
$ pip3 show picamera2
Name: picamera2
Version: 0.2.3
Summary: The libcamera-based Python interface to Raspberry Pi cameras, based on the original Picamera library
Home-page: https://github.com/RaspberryPi/picamera2
Author: Raspberry Pi & Raspberry Pi Foundation
Author-email: picamera2@raspberrypi.com
License: BSD 2-Clause License
Location: /home/pi/.local/lib/python3.9/site-packages
Requires: piexif, v4l2-python3, pillow, PiDNG, python-prctl, pyopengl, numpy, PyQt5, simplejpeg

```

sandyol55, 2022-07-17:
Looks like your libcamera build is out of step with python3-libcamera.

Your python3-libcamera build:is 
v0.0.0+3495-Package: python3-libcamera
Version: 0~git20220705+f30ad033-1

with a libcamera of 
libcamera build: v0.0.0+3495-bd38def2

I think the suitable libcamera build is
 libcamera build: v0.0.0+3700-f30ad033

So perhaps another sudo apt update & sudo apt full-upgrade  cycle needed?

jenyak, 2022-07-17:
@sandyol55 `sudo apt update & sudo apt full-upgrade` have no effect, it says `All packages are up to date`

sandyol55, 2022-07-17:
@jenyak Hi, sorry that didn't help.  I have the same system as you and apart from the libcamera build version I mentioned above everything else shows the same version as yours.  Don't recall doing anything special to get that libcamera 3700 version. (My system is happily running all the latest examples with no errors)

tvoverbeek, 2022-07-17:
The current python3-libcamera version (0~git20220705+f30ad033-1) has as dependency:
libcamera0 (>= 0~git20220705+f30ad033)
The current libcamera0 version is 0~git20220705+f30ad033-1

The output of my libcamera-hello --version is:
```
tvo@picam2-32:~ $ libcamera-hello --version
libcamera-apps build: 35266e84043f 07-07-2022 (17:54:29)
libcamera build: v0.0.0+3700-f30ad033
tvo@picam2-32:~ $
```
@jenyak Your libcamera0 is obviously older.
Do you by chance have an other one in /usr/local/lib (e.g. the one you had to build yourself for the initial picamera2 testing)?

jenyak, 2022-07-17:
Yes, I have. How can I remove it properly and use a new one?

tvoverbeek, 2022-07-17:
@jenyak Delete everything related to libcamera under /usr/local/lib.
The package version is in /usr/lib/arm-linux-gnueabihf.
I do not remember if there is the equivalent /usr/local/lib/arm-linux-gnueabihf. If so, look there.

jenyak, 2022-07-17:
thank you @tvoverbeek
it's work now.

closed 2022-07-17 by jenyak

maurycydziedzic, 2023-11-07:
> The current python3-libcamera version (`0~git20220705+f30ad033-1`) has as dependency: `libcamera0 (>= 0~git20220705+f30ad033)` The current libcamera0 version is 0~git20220705+f30ad033-1
>
> The output [...]

many thanks for that. It also worked in my project.

Sman137, 2024-06-20:
> @jenyak Delete everything related to libcamera under /usr/local/lib. The package version is in /usr/lib/arm-linux-gnueabihf. I do not remember if there is the equivalent /usr/local/lib/arm-linux-gnueabihf. If so, look there.

This worked for me too, thank you : )