raspberrypi/picamera2 #223: Headless Operation Differences

baqwas, 2022-07-18:
Hello,
This note is merely to seek guidance on more efficient use of the **picamera2** library by a newbie.

My _Hello Camera_ code is as follows:

```

myCamera = Picamera2()
if myCamera is None:
    sys.exit("Instantiation failed for Picamera2 ojbect")
config = myCamera.create_still_configuration() # request one set of buffers for hi-res image
if config is None:
    sys.exit("create_still_configuration failed")
myCamera.configure(config)
myCamera.start()          # capture an image
fn = "/home/pi/Pictures/clickit/" + sys.argv[0][:-3] + "_" + str(datetime.datetime.now())[:19].replace(" ", "_") + ".jpg"
myCamera.capture_file(fn) # save image to file
if os.path.isfile(fn):
    if window:
        sendit(fn)
else:
    print(fn + " not captured")
myCamera.stop()           # all done
```

This code runs fine under Thonny with the following output:
```
>>> %Run capture_headless.py
[146:17:52.696348902] [24244]  INFO Camera camera_manager.cpp:293 libcamera v0.0.0+3700-f30ad033
[146:17:52.853577913] [24253]  WARN RPI raspberrypi.cpp:1252 Mismatch between Unicam and CamHelper for embedded data usage!
[146:17:52.854962137] [24253]  INFO RPI raspberrypi.cpp:1368 Registered camera /base/soc/i2c0mux/i2c@1/imx219@10 to Unicam device /dev/media3 and ISP device /dev/media0
[146:17:52.863642276] [24244]  INFO Camera camera.cpp:1029 configuring streams: (0) 3280x2464-BGR888
[146:17:52.864765353] [24253]  INFO RPI raspberrypi.cpp:759 Sensor: /base/soc/i2c0mux/i2c@1/imx219@10 - Selected sensor format: 3280x2464-SBGGR10_1X10 - Selected unicam format: 3280x2464-pBAA
>>> 

```
When I run the program from a terminal window, the following output is displayed and the program appears to be _frozen_:
```

$ ./capture_headless.py
[146:25:44.227469639] [25364]  INFO Camera camera_manager.cpp:293 libcamera v0.0.0+3700-f30ad033
[146:25:44.387544742] [25374]  WARN RPI raspberrypi.cpp:1252 Mismatch between Unicam and CamHelper for embedded data usage!
[146:25:44.388926253] [25374]  INFO RPI raspberrypi.cpp:1368 Registered camera /base/soc/i2c0mux/i2c@1/imx219@10 to Unicam device /dev/media3 and ISP device /dev/media0

```

Presumably, the code (in terminal mode) is stuck at the `myCamera.configure(config)` statement. What am I misunderstanding when attempting to use it from a terminal window? Thanks.

Kind regards.

chrisruk, 2022-07-22:
Hi,

I've had a little look at this, I'm thinking it might be because when you run the capture_headless.py via the terminal you still also have the original python interpreter running.

If I open 2 python interpreters running and run myCamera = Picamera2(), in both of them, the 2nd interpreter will hang for me.

So I think if you close other python interpreters then run capture_headless.py it should work.

I think picamera2 should throw an exception though, rather than hanging in the 2nd terminal, so will have to look into that.

Hope that helps

baqwas, 2022-07-22:
Yes, @chrisruk, you are absolutely correct! How stupid of me not to think about the issue a little deeper before posting. The code was elementary and that kind of _numbed_ me. I connected via SSH and ran the underlying shell script with proper fully qualified folder/filenames. It worked.

I had similar but totally unrelated issue with the previous library running conflicting cron jobs. Thanks.

Kind regards.

closed 2022-07-22 by baqwas

chrisruk, 2022-07-22:
Thanks for reporting this! I think picamera2 should produce an error message rather than hanging in the 2nd instance, as that is rather confusing.  Will create an issue for that.