raspberrypi/picamera2 #226: [HOW-TO] Use the Circular Output?

CricGuru, 2022-07-19:
**I would like to create a python script, which continuously records to a buffer, and then, when a signal is received (say the user types something into the terminal), the buffer is saved, and the previous 5 seconds worth of frames are available to view.**

From the [documentation](https://datasheets.raspberrypi.com/camera/picamera2-manual-draft.pdf) (section 7.2.3), it should be possible. The included example records a clip with 0 frames however, as `output.stop()` is run immediately after `output.start()`. To rectify this I added `time.sleep(5)` between those lines of code. This should record 5 seconds worth of footage after the signal is received.

I have found that if the time between `picam2.start_recording(encoder, output)` and `output.start()` exceeds the buffersize (default 150 frames) then the output file has some issues with it - VLC does not play the file, MP4Box does not accept the file, but the file still has a size in the order of MiBs. I do not know much about video files, but I suspect the output file loses some key information when the old frames are rewritten by the new ones.

Here is my code:

```
from picamera2.encoders import H264Encoder
from picamera2.outputs import CircularOutput
from picamera2 import Picamera2
import time
import sys
picam2 = Picamera2()
picam2.configure(picam2.create_video_configuration())
encoder = H264Encoder()
output = CircularOutput(buffersize = 150)
picam2.start_recording(encoder, output)
t = input()
# Now when it's time to start recording the output, including the previous 5 seconds:
output.fileoutput = "file.h264"
output.start()
time.sleep(5)
# And later it can be stopped with:
output.stop()
```

chrisruk, 2022-07-20:
Thanks for reporting this, just looking into the reason behind this issue

chrisruk, 2022-08-15:
@CricGuru Hi, I've created a new example, which I think is similar to what you wanted to do, in the 'next' branch of this repo.
This should get merged into 'main' when picamera2 is next released.

https://github.com/raspberrypi/picamera2/blob/next/examples/capture_circular_nooutput.py

I added a new parameter to CircularOutput, 'outputtofile' which you set to false, so that frames are only written from the circular buffer when .stop() is called.

chrisruk, 2022-08-24:
The changes for this are now in main.  Please re-open if there's any remaining issue.

closed 2022-08-24 by chrisruk