raspberrypi/picamera2 #229: [BUG] Hanging at constructor, rather than throwing exception when 2+ interpreters try to initialise picamera2

chrisruk, 2022-07-22:
**Describe the bug**

If you run two interpreters both with: 
p = Picamera2()

The second interpreter will hang on this call.

**Expected behaviour**

An exception should be produced in the second call to picamera2.

davidplowman, 2022-08-10:
I'm suspicious that this may actually be a libcamera bug. The 2nd instance realises that it can't lock all the media devices, and therefore decides to unlock them all again and give up: [here](https://git.linuxtv.org/libcamera.git/tree/src/libcamera/pipeline_handler.cpp#n167). 

Only that very call to `unlock()` causes it to try to re-acquire the mutex (`lock_`) that it already has, causing it to hang. Not sure what the best fix is there... give up `lock_` first? have a version of `unlock()` that doesn't re-take the lock? make `lock_` itself a re-entrant kind of a mutex?

@kbingham Does this sound right? Any thoughts on the best course of action?

kbingham, 2022-08-10:
Could you try testing with this patch please? https://patchwork.libcamera.org/patch/16728/

It sounds like something I reported on IPU3 recently, and Laurent posted that as a fix.

davidplowman, 2022-08-10:
Ah yes, that does indeed fix it for me! I see the "unlocking" bit that happens when it fails is now removed from being under the mutex.

So what's the ETA for this one...?   Thanks!!

kbingham, 2022-08-10:
Could you reply to the patch saying it fixes an issue for you please? (+RB/TB tags :D)

It fixes the IPU3 for me - but breaks CTS (chrome camera test suite) because now it lets two cameras run, and that breaks - but that's not /your/ problem. So I'm holding up the patch ... but if it solves a problem for you we should look to merge it.

kbingham, 2022-08-10:
Now merged at https://git.libcamera.org/libcamera/libcamera.git/commit/?id=dfc6d711c9f7f0a9868afa5158aa2089163bded3

davidplowman, 2022-08-11:
@kbingham Fabulous, thank you!

I'm going to close this one as it will go away when we pick up the latest libcamera for our next release. Thanks everyone!

closed 2022-08-11 by davidplowman