raspberrypi/picamera2 #231: [BUG]

DrMicio, 2022-07-25:
**Describe the bug**
I can't run any python script from other users other than pi
I have 2 different users (pi and irontotem) and with the second user it doesn't work
libcamera commands work
running the same script with sudo doesn't work either

**Console Output, Screenshots**
This is the console output

```
python3 /opt/irontotem/scripts/camera_server.py
[0:02:54.919448416] [1051]  INFO Camera camera_manager.cpp:293 libcamera v0.0.0+3700-f30ad033
[0:02:54.957610196] [1052]  WARN RPI raspberrypi.cpp:1252 Mismatch between Unicam and CamHelper for embedded data usage!
[0:02:54.958273367] [1052]  INFO RPI raspberrypi.cpp:1368 Registered camera /base/soc/i2c0mux/i2c@1/imx219@10 to Unicam device /dev/media3 and ISP device /dev/media1
2022-07-25T14:41:00.351Z | ERROR    | Camera __init__ sequence did not complete.
Traceback (most recent call last):
  File "/home/irontotem/.local/lib/python3.9/site-packages/picamera2/picamera2.py", line 77, in __init__
    self.open_camera()
  File "/home/irontotem/.local/lib/python3.9/site-packages/picamera2/picamera2.py", line 178, in open_camera
    if self.initialize_camera():
  File "/home/irontotem/.local/lib/python3.9/site-packages/picamera2/picamera2.py", line 162, in initialize_camera
    self.sensor_resolution = self.camera.properties["PixelArraySize"]
KeyError: 'PixelArraySize'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/opt/irontotem/scripts/camera_server.py", line 59, in <module>
    picam2 = Picamera2(tuning=tuning)
  File "/home/irontotem/.local/lib/python3.9/site-packages/picamera2/picamera2.py", line 81, in __init__
    raise RuntimeError("Camera __init__ sequence did not complete.")
RuntimeError: Camera __init__ sequence did not complete.
```

**Hardware :**
Raspberry Pi4 4 GB RAM
```
uname -a
Linux raspberrypi 5.15.32-v7l+ #1538 SMP Thu Mar 31 19:39:41 BST 2022 armv7l GNU/Linux

```

TimBrownConsulting, 2022-07-26:
I got this error when I had two versions of libcamera installed, and was accidentally trying to use the older version.  Is it possible that you installed the newest version under the pi account and the irontotem account is using the original distribution version?

davidplowman, 2022-07-26:
Yes, this is usually down to mismatching versions. Probably best to ensure both libcamera and picamera2 are updated with
```
sudo apt install -y python3-libcamera
pip3 install picamera2 --upgrade
```
and report back. Thanks!

DrMicio, 2022-07-26:
Yes, that was it. Thank you so much for your help!

closed 2022-07-26 by DrMicio