raspberrypi/picamera2 #232: [HOW-TO] Set auto white balance mode with camera controls

CorvetteCole, 2022-07-25:
**Describe what it is that you want to accomplish**
Set auto white balance mode

**Describe alternatives you've considered**
N/A

**Additional context**
I've dug through the code, but neither `_VIRTUAL_FIELDS_MAP_` in controls.py or `camera_ctrl_info` in picamera2.py has illuminated this information for me. It is implied that this would be controlled via the camera controls interface on page 21 of the documentation (along with exposure, sharpness, etc), but I cannot figure out how to actually use this.

I understand this library is incomplete so if this is simply not implemented yet just let me know

closed 2022-07-25 by CorvetteCole

flatsiedatsie, 2022-08-03:
Could you explain what you learnt? Is it possible?

CorvetteCole, 2022-08-03:
@flatsiedatsie some further information on how to use this can be found in this issue: https://github.com/raspberrypi/picamera2/issues/212.

basically, the controls available are actually libcamera controls and are more accurately referred to there. so for different AwbMode values, take a look at the libcamera documentation that I linked above

flatsiedatsie, 2022-08-03:
Thanks. I'm trying to switch between `normal` and 'night mode' on the fly. With PiCamera I could use:
```
self.picam.exposure_mode = "auto"
self.picam.image_effect = "none"
self.picam.exposure_compensation = 0
self.picam.ISO = 0
self.picam.brightness = 50
self.picam.contrast = 0
```
and
```
self.picam.exposure_mode = "night"
self.picam.image_effect = "denoise"
self.picam.exposure_compensation = 25
self.picam.ISO = 800
self.picam.brightness = 70
self.picam.contrast = 50
```

With PiCamera2 I was trying `self.picam.set_controls({"ExposureTime": 100000, "AnalogueGain": 1})`. I'll see if I can get close to what I had.

CorvetteCole, 2022-08-26:
you might need to turn off auto exposure for the exposuretime to work. just a thought