raspberrypi/picamera2 #238: [SUGGESTION] Inconsistency between British and International English.

eat-sleep-code, 2022-07-27:
**Please only include one suggestion or features request per issue!**

**Is your feature request related to a problem? Please describe.**
No, it is related to an inconsistency in spelling.

**Describe the solution you'd like**
I note that the very-British spelling of colour was used.    But this results in an inconsistency:

`configPreview.colour_space = ColorSpace.Jpeg()`

**Describe alternatives you've considered**
I mean I can live with it.   :-)

**Additional context**

davidplowman, 2022-07-28:
Indeed, `colour_space` is because I always default to British English (seeing as I am British), but `ColorSpace` comes from libcamera which I believe is meant to use International (American) English, though I wouldn't like to guarantee it's entirely consistent everywhere. I'm not particularly inclined to move the whole of _Picamera2_ to American spellings so I don't think this is ever going to get resolved...  :(

davidplowman, 2022-07-28:
I think I'll close this one if that's OK (sorry)

closed 2022-07-28 by davidplowman